apache/nuttx #9575: Potential stale memory access in VFS

pkarashchenko, 2023-06-19:
I've noticed that `fs_getfilep()` returns the pointer from `list->fl_files`. The returned pointer is used further and passed to other APIs. The `list->fl_files` access is protected by `list->fl_lock`, but after `fs_getfilep()` returns the `list->fl_lock` is released. On the other hand the `files_extend()` contains `tmp = kmm_realloc(list->fl_files, sizeof(FAR struct file *) * row);` call that potentially can move all the list entries to a new memory location. So it one thread is interrupted after `fs_getfilep()` returns and another thread leads to `files_extend()` call the first thread will potentially access already freed memory (maybe even allocated and used by another user), so potentially can cause memory corruption.

xiaoxiang781216, 2023-06-19:
it isn't an issue since the reallocation point to struct file *, not struct file.

pkarashchenko, 2023-06-19:
> it isn't an issue since the reallocation point to struct file *, not struct file.

Indeed. The reference to struct file is stored in the list and remains unchanged

closed 2023-06-19 by pkarashchenko