apache/nuttx #9596: Failure to build misc/lib_memfd.c

fdcavalcanti, 2023-06-22:
I just rebased my current fork and can't compile anymore. Something to do with misc/lib_memfd.c.
Here's the information:

```
misc/lib_memfd.c: In function 'memfd_create':
CC:  misc/lib_crc32.c misc/lib_memfd.c:39:32: error: 'CONFIG_LIBC_MEM_FD_VFS_PATH' undeclared (first use in this function); did you mean 'CONFIG_LIBC_FTOK_VFS_PATH'?
   39 | #  define LIBC_MEM_FD_VFS_PATH CONFIG_LIBC_MEM_FD_VFS_PATH
      |                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~
misc/lib_memfd.c:42:34: note: in expansion of macro 'LIBC_MEM_FD_VFS_PATH'
   42 | #define LIBC_MEM_FD_VFS_PATH_FMT LIBC_MEM_FD_VFS_PATH "/%s"
      |                                  ^~~~~~~~~~~~~~~~~~~~
misc/lib_memfd.c:56:32: note: in expansion of macro 'LIBC_MEM_FD_VFS_PATH_FMT'
   56 |   snprintf(path, sizeof(path), LIBC_MEM_FD_VFS_PATH_FMT, name);
      |                                ^~~~~~~~~~~~~~~~~~~~~~~~
misc/lib_memfd.c:39:32: note: each undeclared identifier is reported only once for each function it appears in
   39 | #  define LIBC_MEM_FD_VFS_PATH CONFIG_LIBC_MEM_FD_VFS_PATH
      |                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~
misc/lib_memfd.c:42:34: note: in expansion of macro 'LIBC_MEM_FD_VFS_PATH'
   42 | #define LIBC_MEM_FD_VFS_PATH_FMT LIBC_MEM_FD_VFS_PATH "/%s"
      |                                  ^~~~~~~~~~~~~~~~~~~~
misc/lib_memfd.c:56:32: note: in expansion of macro 'LIBC_MEM_FD_VFS_PATH_FMT'
   56 |   snprintf(path, sizeof(path), LIBC_MEM_FD_VFS_PATH_FMT, name);
      |                                ^~~~~~~~~~~~~~~~~~~~~~~~
misc/lib_memfd.c:42:55: error: expected ')' before string constant
   42 | #define LIBC_MEM_FD_VFS_PATH_FMT LIBC_MEM_FD_VFS_PATH "/%s"
      |                                                       ^~~~~
misc/lib_memfd.c:56:32: note: in expansion of macro 'LIBC_MEM_FD_VFS_PATH_FMT'
   56 |   snprintf(path, sizeof(path), LIBC_MEM_FD_VFS_PATH_FMT, name);
      |                                ^~~~~~~~~~~~~~~~~~~~~~~~
make[1]: *** [Makefile:150: bin/lib_memfd.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [tools/LibTargets.mk:180: libs/libc/libc.a] Error 2
make: *** Waiting for unfinished jobs....

```

Maybe someone who worked on it could check?

Thanks

pkarashchenko, 2023-06-22:
Have you tried to run `make menuconfig` and save `defconfig`? I think that you need to refresh your configuration file after https://github.com/apache/nuttx/pull/9513

fdcavalcanti, 2023-06-22:
Thanks for the info. That fixed it.

closed 2023-06-22 by fdcavalcanti