apache/nuttx #9598: Suggestion: Add "Description:" header to defconfig file

acassis, 2023-06-23:
Currently we use the name of the board profile directory to give some clue to users what is the purpose of that defconfig, i.e.: nsh, usbnsh, netnsh, timer, kostest, etc.

But in some case it is not enough, for people familiar to NuttX will know that is kostest, but for a newcomer it doesn't make much sense.

Also now on ESP32S3 we will have to USB Console options: USB Console using USBJTAGSERIAL peripheral (that we will call as usnconsole) and USB Console using USB OTG Device (that we call usbnsh).

btashton, 2023-06-24:
This information should be in the board documentation (either the docs or the readme for now). This is just delaying that work.

acassis, 2023-06-24:
Hi @btashton I totally agree with it, but it is hard to convince people to create a documentation entry for new board profiles. Question: is it possible to create a CI rules to verify if the PR is creating a new defconfig and if so generate an error case the commit is not create an Documentation/ entry?

cederom, 2023-06-25:
Just let me know what to put and where and I will help.. lets keep things coherent and make folks used to the process :-)