apache/nuttx #9602: nuttx/include/arch/irq.h:700: Error: unrecognized opcode `csrw mstatus,s0', extension `zicsr' required

TaiJuWu, 2023-06-24:
I try to build rv-virt:smp64
```
./tools/configure.sh rv-virt:smp64
```

When compile codes, show this error:
nuttxspace/nuttx/include/arch/irq.h:700: Error: unrecognized opcode `csrw mstatus,s0', extension `zicsr' required

toolchain: riscv64-unknown-elf-gcc (g2ee5e430018) 12.2.0

acassis, 2023-06-24:
@masayuki2009 have you seen this error before?

masayuki2009, 2023-06-24:
@TaiJuWu 
Could you tell us what compiler you are using now?
In my understanding, the latest RISC-V compiler needs some extensions.

TaiJuWu, 2023-06-24:
Im using ```riscv64-unknown-elf-gcc (g2ee5e430018) 12.2.0```

masayuki2009, 2023-06-24:
>Im using riscv64-unknown-elf-gcc (g2ee5e430018) 12.2.0

https://xpack.github.io/blog/2022/05/18/riscv-none-elf-gcc-v12-1-0-2-released/

```
RISC-V ISA updates

Compared to previous releases, starting from 12.x, the compiler implements the new RISC-V ISA, which introduces an incompatibility issue, and builds might throw error messages like unrecognized opcode csrr.

The reason is that csr read/write (csrr*/csrw*) instructions and fence.i instruction were separated from the I extension, becoming two standalone extensions: Zicsr and Zifencei.

The solution is to add _zicsr and/or _zifencei to the -march option, e.g. -march=rv32imac becomes -march=rv32imac_zicsr_zifencei.
```

TaiJuWu, 2023-06-24:
I try to compile code and the compiler option is

```
riscv64-unknown-elf-gcc -MT ./clock_settime.o  -M '-fno-common' '-Wall' '-Wstrict-prototypes' '-Wshadow' '-Wundef' '-Os' '-fno-strict-aliasing' '-fomit-frame-pointer' '-ffunction-sections' '-fdata-sections' '-g' '-march=rv64imac' '-mabi=lp64' '-mcmodel=medany' '-isystem' '/home/tjwu1217/nuttxspace/nuttx/include' '-D__NuttX__' '-DNDEBUG' '-D__KERNEL__' '-pipe' '-I' '/home/tjwu1217/nuttxspace/nuttx/sched' '-march=rv64imac_zicsr_zifencei' ./clock/clock_settime.c
```
I try to use grep to find rv64imac in all repo but  it is not exist in this repo.
Could you tell me where should I modify this option?

acassis, 2023-06-25:
@TaiJuWu did you try to edit arch/risc-v/src/common/Toolchain.defs ?

TaiJuWu, 2023-06-26:
@acassis Thank you for your reply.
I've compiled the code successfully with your suggestion.

closed 2023-06-26 by TaiJuWu