apache/nuttx #9616: Implicit delaration of memset when building against Nuttx export.

g2gps, 2023-06-26:
I'm building a third party library against Nuttx export, with CONFIG_BUILD_KERNEL. 

When building:

```c
int test(int fd) {
    fd_set fds;
    int r;

    FD_ZERO(&fds);
    FD_SET(fd, &fds);

    if ((r = select(fd+1, NULL, &fds, NULL, NULL)) < 0)
        return -1;

    assert(r > 0);

    return 0;
}
```

I receive the warning:
```
test.c: In function 'test':
nuttx/include/sys/select.h:95:4: warning: implicit declaration of function 'memset' [-Wimplicit-function-declaration]
   95 |    memset((set), 0, sizeof(fd_set))
      |    ^~~~~~
test.c:63:5: note: in expansion of macro 'FD_ZERO'
   63 |     FD_ZERO(&fds);
      |     ^~~~~~~
test.c:30:1: note: include '<string.h>' or provide a declaration of 'memset'
   29 | #include "test.h"
  +++ |+#include <string.h>
   30 | 
```

But interestingly, when I modify an existing Nuttx example, in this case "hello", like this:

```diff
diff --git a/examples/hello/hello_main.c b/examples/hello/hello_main.c
index 2cd6cb400..a3635f218 100644
--- a/examples/hello/hello_main.c
+++ b/examples/hello/hello_main.c
@@ -22,8 +22,7 @@
  * Included Files
  ****************************************************************************/
 
-#include <nuttx/config.h>
-#include <stdio.h>
+#include <sys/select.h>
 
 /****************************************************************************
  * Public Functions
@@ -35,6 +34,9 @@
 
 int main(int argc, FAR char *argv[])
 {
-  printf("Hello, World!!\n");
+  fd_set fds;
+  FD_ZERO(&fds);
+
+  /* Just return something for the example */
   return 0;
 }

```

I don't receive any warning. 

I'm building hello_main.c with:
```bash
riscv64-unknown-elf-gcc -c -fno-common -Wall -Wstrict-prototypes -Wshadow -Wundef -fno-common -pipe  -Os -fno-strict-aliasing -fomit-frame-pointer -march=rv32imac -mabi=ilp32 -isystem apps/import/include -isystem apps/import/include -D__NuttX__ -DNDEBUG  -I "apps/include"   hello_main.c -o  apps.examples.hello.o
```

The resulting Make.dep for hello_main is:
```bash
hello_main.c.apps.examples.hello.o: \
hello_main.c \
apps/import/include/sys/select.h \
apps/import/include/nuttx/config.h \
apps/import/include/limits.h \
apps/import/include/arch/limits.h \
apps/import/include/stdint.h \
apps/import/include/nuttx/compiler.h \
apps/import/include/arch/types.h \
apps/import/include/arch/inttypes.h \
apps/import/include/signal.h \
apps/import/include/time.h \
apps/import/include/sys/types.h \
apps/import/include/sys/time.h
# No files specified for dependency generation
# No files specified for dependency generation
``` 

I may be missing something obvious, but I can't see where the definition for `memset` is coming from? 

Naturally, the issue could be fixed by adding `#include <string.h>` to `sys/select.h`, but it would be nice to understand why the same warning doesn't occur when building using Nuttx's `make import` from the apps directory.

mu578, 2023-06-27:
FD_ZERO should by defined as explicit_bzero and having select.h including string

xiaoxiang781216, 2023-06-27:
@g2gps the simplest method is:

1. make V=1 to get the command line
2. Append -E to the command line and run again
3. Compare the generated file to see the difference

closed 2023-06-27 by pkarashchenko