apache/nuttx #9624: Support to save crash dump in the flash

acassis, 2023-06-26:
Currently there is not a simple way to save crash dump in the flash.

This information (crash dump) could be lost on systems running in the field with watchdog enabled.

Related issue: https://github.com/apache/nuttx/issues/3417

Ref: https://cwiki.apache.org/confluence/display/NUTTX/Debug+Output+%28SYSLOG%29+Issues

xiaoxiang781216, 2023-06-26:
You can try:
https://github.com/apache/nuttx/blob/master/binfmt/binfmt_coredump.c#L46
with:
https://github.com/apache/nuttx/blob/master/libs/libc/stream/lib_mtdoutstream.c
to save the core dump to flash and follow https://github.com/apache/nuttx/pull/4871
or with:
https://github.com/apache/nuttx/blob/master/sched/misc/assert.c#L642-L646
to sieral and follow the https://github.com/apache/nuttx/pull/9369

acassis, 2023-06-26:
Thank you very much @xiaoxiang781216 I'll try it and write a Guide in our documentation

xiaoxiang781216, 2023-06-27:
BTW, you can even get the local variables with the default panic log by minidump server: https://github.com/apache/nuttx/pull/5836