apache/nuttx #9642: lib_rexec has big security hole

szsam, 2023-06-28:
According to the man page of rexec, the rexec() function sends the unencrypted password across the network. This is a big security hole. Shall we disable this library, or at least give users a security warning?

acassis, 2023-06-29:
@xiaoxiang781216 @pkarashchenko @btashton suggestion?

btashton, 2023-06-29:
I don't have a strong opinion on this, but my inclination is to leave it be since it is not something that will be in the build unless someone actually uses it. There are many utilities that are not secure if the network transport is not secure. Just like someone using plain ftp.

xiaoxiang781216, 2023-06-29:
rexec/rexec_af just a passive function which doesn't have any side effect if no program calls them. The applications which expose remote execution functionality are:
apps/netutils/rexec
apps/netutils/rexecd
Both are disable by default and require the user enable them explicitly in defconfig.

xiaoxiang781216, 2023-06-29:
BTW, exec/execd developed by @Donny9 is really used to execute a program on remote CPU from the main CPU in one SoC, which is perfect to ignore the security issue.

mu578, 2023-06-29:
@xiaoxiang781216  BSD or Linux like implementation behave the same since ever...
 
```From a basic linux distro manual:

Security
Users of this command should be aware that [rexec] transmits their password to the remote host clear text, not encrypted. If the network is not secure to the remote host, the password can be comprimised.
```

So the text just states if within a case your network is not secure "be aware that [rexec] doesn't nothing much than just transmit data to remote" ; thus just read; "your responsibility to get it right."

closed 2023-07-01 by xiaoxiang781216