apache/nuttx #9645: how to debug elf

xiaotailang, 2023-06-29:
Hello everyone,

I have compiled my application into an ELF file, for example, "myapp_elf", following the instructions in this [issue.](https://github.com/apache/nuttx/issues/2323), After downloading it to the development board, I want to debug this "myapp_elf" file. What configuration options should I add to support debugging ELF files? I have referred to this link for debugging [methods,](https://cwiki.apache.org/confluence/display/NUTTX/Debugging+ELF+Loadable+Modules) but it is not very convenient to use and sometimes results in errors. Are there any other better alternatives?

xiaotailang, 2023-07-03:
@acassis okay thank you very much !

xiaotailang, 2023-07-18:
Hi @acassis I followed the previous advice and enabled the debug configuration. After tracing, I discovered the issue causing my ELF file to fail loading. It was because I used global variables in my code, and the compilation options lacked "-fno-common". Later, when I used GDB to debug the ELF file "myapp_elf", I found that in most cases, breakpoints couldn't be triggered, and the "step" command couldn't enter function internals. However, I'm not sure if it's a problem with GDB or the compiled ELF file itself. The compilation options I used were "-Os" and "-g". I also tried the "-O0" and "-g" options, but if I compile the ELF with the "-O0" option, it results in an error when loading the ELF. Is there a complete method for debugging ELF files using GDB?

yuening1999, 2024-10-25:
> Hi @acassis I followed the previous advice and enabled the debug configuration. After tracing, I discovered the issue causing my ELF file to fail loading. It was because I used global variables in [...]

Hi, [xiaotailang](https://github.com/xiaotailang), Do you resolve the trouble?

I also met the similar problem, when gdb debugging my app, I found the logic of my app is not correct.  Local variable for "for loop"on my app is error value and random.
meanwhile, breakpoint and "s" command both work fine.  But I download my app to file system on board, my app run normally.

I also want to know correct method to debug external elf program.