apache/nuttx #9670: Can't compile sim:citest with CONFIG_HOST_MACOS=y

pkarashchenko, 2023-07-02:
Steps to reproduce:
1. `./tools/configure.sh sim:citest`
2. `make menuconfig` -> select `CONFIG_LIBCXXABI=y`
3. `make -j8` ->
```
Create version.h
sed: 1: "/CONFIG_BASE_DEFCONFIG/ ...": bad flag in substitute command: '}'
make: *** [include/nuttx/config.h] Error 1
make: *** Waiting for unfinished jobs....
```

pkarashchenko, 2023-07-02:
Seems to be regression after https://github.com/apache/nuttx/commit/bc4266fa485

pkarashchenko, 2023-07-02:
@lucasssvaz could you please take a look?

acassis, 2023-07-02:
@lucasssvaz please take a look!

closed 2023-07-03 by acassis

lucasssvaz, 2023-07-10:
Just came back from vacation, sorry for all the trouble.
I messed up the differences between Linux and OSX sed's. You can't omit the semicolon on OSX, that's why the error occurs.
@pkarashchenko I currently don't have an OSX machine to test. Could you please check if #9754 now works as expected ?

pkarashchenko, 2023-07-10:
Ok. Let me check

pkarashchenko, 2023-07-10:
@lucasssvaz seems to be working fine.