apache/nuttx #9674: setvbuf test in ostest fails on MACOS with 'Illegal instruction: 4'

pkarashchenko, 2023-07-02:
Steps to reproduce:
1. `./tools/configure.sh sim:ostest`
2. `make -j8`
3. `./nuttx`
4. Observe next output
```
$ ./nuttx
stdio_test: write fd=1
stdio_test: Standard I/O Check: printf
stdio_test: write fd=2
stdio_test: Standard I/O Check: fprintf to stderr
ostest_main: putenv(Variable1=BadValue3)
ostest_main: setenv(Variable1, GoodValue1, TRUE)
ostest_main: setenv(Variable2, BadValue1, FALSE)
ostest_main: setenv(Variable2, GoodValue2, TRUE)
ostest_main: setenv(Variable3, GoodValue3, FALSE)
ostest_main: setenv(Variable3, BadValue2, FALSE)
show_variable: Variable=Variable1 has value=GoodValue1
show_variable: Variable=Variable2 has value=GoodValue2
show_variable: Variable=Variable3 has value=GoodValue3
ostest_main: Started user_main at PID=4

user_main: Begin argument test
user_main: Started with argc=5
user_main: argv[0]="ostest"
user_main: argv[1]="Arg1"
user_main: argv[2]="Arg2"
user_main: argv[3]="Arg3"
user_main: argv[4]="Arg4"

End of test memory usage:
VARIABLE  BEFORE   AFTER
======== ======== ========
arena     3efffe8  3efffe8
ordblks         2        2
mxordblk  3eb98f0  3eb98f0
uordblks    466c8    466c8
fordblks  3eb9920  3eb9920

user_main: getopt() test
getopt():  Simple test
getopt():  Invalid argument
getopt():  Missing optional argument
getopt_long():  Simple test
getopt_long():  No short options
getopt_long():  Argument for --option=argument
getopt_long():  Invalid long option
getopt_long():  Mixed long and short options
getopt_long():  Invalid short option
getopt_long():  Missing optional arguments
getopt_long_only():  Mixed long and short options
getopt_long_only():  Single hyphen long options

End of test memory usage:
VARIABLE  BEFORE   AFTER
======== ======== ========
arena     3efffe8  3efffe8
ordblks         2        2
mxordblk  3eb98f0  3eb98f0
uordblks    466c8    466c8
fordblks  3eb9920  3eb9920
show_variable: Variable=Variable1 has value=GoodValue1
show_variable: Variable=Variable2 has value=GoodValue2
show_variable: Variable=Variable3 has value=GoodValue3
show_variable: Variable=Variable1 has no value
show_variable: Variable=Variable2 has value=GoodValue2
show_variable: Variable=Variable3 has value=GoodValue3

End of test memory usage:
VARIABLE  BEFORE   AFTER
======== ======== ========
arena     3efffe8  3efffe8
ordblks         2        3
mxordblk  3eb98f0  3eb98f0
uordblks    466c8    466a8
fordblks  3eb9920  3eb9940
show_variable: Variable=Variable1 has no value
show_variable: Variable=Variable2 has no value
show_variable: Variable=Variable3 has no value

End of test memory usage:
VARIABLE  BEFORE   AFTER
======== ======== ========
arena     3efffe8  3efffe8
ordblks         3        3
mxordblk  3eb98f0  3eb98f0
uordblks    466a8    46618
fordblks  3eb9940  3eb99d0
tls: Successfully set 0
tls: Successfully set ffffffff
tls: Successfully set 55555555
tls: Successfully set aaaaaaaa

End of test memory usage:
VARIABLE  BEFORE   AFTER
======== ======== ========
arena     3efffe8  3efffe8
ordblks         3        3
mxordblk  3eb98f0  3eb98f0
uordblks    46618    46618
fordblks  3eb99d0  3eb99d0

user_main: setvbuf test
setvbuf_test: Test NO buffering
Illegal instruction: 4
```

acassis, 2023-07-04:
@pkarashchenko Did you try an old version?

pkarashchenko, 2023-07-04:
Old version of MAC OS or older version of the code?

pkarashchenko, 2023-07-04:
Currently tested only with NuttX mainline and MAC OS 13.4.1

pkarashchenko, 2023-07-04:
Just found that line `stream->fs_bufend   = newbuf + size;` leads to `Illegal instruction: 4` if `newbuf` is `NULL` and `size` is `0`.

pkarashchenko, 2023-07-04:
The `size` value does not matter. It is reproduced with `-fsanitize=undefined` option when `NULL` pointer participate in arithmetic operations (add or subtract).

pkarashchenko, 2023-07-04:
If I replace `-fsanitize=undefined` with `-fsanitize=pointer-overflow`, I'm getting
```
user_main: setvbuf test
setvbuf_test: Test NO buffering
ubsan_prologue: ================================================================================
ubsan_prologue: UBSAN: pointer-overflow in stdio/lib_setvbuf.c:243:32
ubsan_epilogue: ================================================================================
setvbuf_test: Using NO buffering
```

pkarashchenko, 2023-07-04:
Seems to be caused by `CONFIG_MM_UBSAN_TRAP_ON_ERROR=y`

pkarashchenko, 2023-07-04:
Clang:
```
-fsanitize=pointer-overflow: Performing pointer arithmetic which overflows, or where either the old or new pointer value is a null pointer (or in C, when they both are).
```
GCC:
```
-fsanitize=pointer-overflow
This option enables instrumentation of pointer arithmetics. If the pointer arithmetics overflows, a run-time error is issued.
```

xiaoxiang781216, 2023-07-04:
It's better to avoid the math operation with NULL pointer.

pkarashchenko, 2023-07-04:
Let me see if I can rework code with minimal performance impact

pkarashchenko, 2023-07-04:
We can try to go with `stream->fs_bufend   = newbuf != NULL ? newbuf + size : NULL;`, but that is a bit ugly.