apache/nuttx #9688: warning: ‘%s’ directive output may be truncated writing up to 107 bytes into a region of size 100

acassis, 2023-07-03:
@anchao I think the issue you fixed on 8e146590a9dccec39752f1d2ee094c87508d9635 still happening:

```
CC:  local/local_fifo.c In file included from local/local_fifo.c:25:
In function ‘local_format_name’,
    inlined from ‘local_hd_name’ at local/local_fifo.c:132:3,
    inlined from ‘local_open_receiver’ at local/local_fifo.c:661:3:
local/local_fifo.c:77:16: warning: ‘%s’ directive output may be truncated writing up to 107 bytes into a region of size 100 [-Wformat-truncation=]
   77 |                CONFIG_NET_LOCAL_VFS_PATH "/%s%s", inpath, suffix);
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~
local/local_fifo.c: In function ‘local_open_receiver’:
local/local_fifo.c:82:44: note: format string is defined here
   82 |                CONFIG_NET_LOCAL_VFS_PATH "/%s%s%" PRIx32,
      |                                            ^~
In function ‘local_format_name’,
    inlined from ‘local_hd_name’ at local/local_fifo.c:132:3,
    inlined from ‘local_open_receiver’ at local/local_fifo.c:661:3:
local/local_fifo.c:76:7: note: ‘snprintf’ output between 12 and 119 bytes into a destination of size 109
   76 |       snprintf(outpath, LOCAL_FULLPATH_LEN - 1,
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   77 |                CONFIG_NET_LOCAL_VFS_PATH "/%s%s", inpath, suffix);
      |                ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
LD:  nuttx

local/local_fifo.c:76:7: note: ‘snprintf’ output between 12 and 119 bytes into a destination of size 109
```

acassis, 2023-07-04:
I tested using sim:sotest but I think it will happen with other configs as well.

anchao, 2023-07-06:
@acassis Thank you, I send a PR to fix this warning, please review PR https://github.com/apache/nuttx/pull/9714

closed 2023-07-06 by xiaoxiang781216