apache/nuttx #9695: Deadloop in VFS if CONFIG_CANCELLATION_POINTS is enabled

pkarashchenko, 2023-07-04:
I'm having a system built in FLAT mode with `CONFIG_CANCELLATION_POINTS=y`. The system is running multiple tasks and one task is sending messages over CAN. There is send frame procedure that is opening a file descriptor -> calling write -> closing file descriptor. Sometimes that tasks is hitting a dead loop if it is canceled. The execution stocks in `open` call:
`open()` -> `enter_cancellation_point()` -> `nx_vopen()` -> `file_vopen()` -> `inode_release()` -> `inode_lock()` -> `nxrmutex_lock()` -> `nxmutex_lock()` -> `_SEM_WAIT()/sem_wait() that is evaluated to sem_wait() because of #if !defined(CONFIG_BUILD_FLAT) && defined(__KERNEL__)` -> 
```
      do
        {
          ret = inode_lock();

          /* This only possible error is due to cancellation of the thread.
           * We need to try again anyway in this case, otherwise the
           * reference count would be wrong.
           */

          DEBUGASSERT(ret == OK || ret == -ECANCELED);
        }
      while (ret < 0);
```
and 
```
int sem_wait(FAR sem_t *sem)
{
  int errcode;
  int ret;

  if (sem == NULL)
    {
      set_errno(EINVAL);
      return ERROR;
    }

  /* sem_wait() is a cancellation point */

  if (enter_cancellation_point())
    {
#ifdef CONFIG_CANCELLATION_POINTS
      /* If there is a pending cancellation, then do not perform
       * the wait.  Exit now with ECANCELED.
       */

      errcode = ECANCELED;
      goto errout_with_cancelpt;
#endif
    }
```
So the nested cancellation point happens that `sem_wait()` return `ECANCELED` to `nxmutex_lock()` that is implemented as
```
int nxmutex_lock(FAR mutex_t *mutex)
{
  int ret;

  DEBUGASSERT(!nxmutex_is_hold(mutex));
  for (; ; )
    {
      /* Take the semaphore (perhaps waiting) */

      ret = _SEM_WAIT(&mutex->sem);
      if (ret >= 0)
        {
          mutex->holder = _SCHED_GETTID();
          break;
        }

      ret = _SEM_ERRVAL(ret);
      if (ret != -EINTR && ret != -ECANCELED)
        {
          break;
        }
    }

  return ret;
}
```
so task spinns forever in `nxmutex_lock()`.

pkarashchenko, 2023-07-04:
Related to https://github.com/apache/nuttx/pull/6338

pkarashchenko, 2023-07-04:
@patacongo I would appreciate your advice here

patacongo, 2023-07-04:
Hi, Petro!  Clearly the problem is either with (1) in this loop termination condition:

      ret = _SEM_ERRVAL(ret);
      if (ret != -EINTR && ret != -ECANCELED)
        {
          break;
        }

Or (2) with the return value from __SEM_WAIT().

The first would happen if the thread has call pthread_mutex_lock() and is canceled while waiting.  That is a normal event.  pthread_mutex_lock() is not a cancellation point, oddly.  https://pubs.opengroup.org/onlinepubs/009695399/functions/xsh_chap02_09.html .  So the looping behavior is probably normal.

The only odd thing I see is that -ECANCELED should be reported only once.  It does not latch (at least, in the past, it did not latch.  I would need to double check the current behavior).  So -ECANCELED should be received once and should not cause an infinite loop.  Similarly with -EINTR; it does not latch either.

If __SEM_WAIT were reporting -ECANCELED multiple times, that would be an error and think a lot things would be broken.

So, it seems that __SEM_WAIT() is returning an error???  __SEM_WAIT() does not often fail so it would be interesting to know if it is returning an error and what the error is.

pkarashchenko, 2023-07-04:
The `-ECANCELED` is returned always because of nested cancelation point. The most top `enter_cancellation_point()` is called from `open()`, so all later calls of cancelable APIs from inside the `open()` will be a nested cancelation points:
```
  /* sem_wait() is a cancellation point */

  if (enter_cancellation_point())
    {
#ifdef CONFIG_CANCELLATION_POINTS
      /* If there is a pending cancellation, then do not perform
       * the wait.  Exit now with ECANCELED.
       */

      errcode = ECANCELED;
      goto errout_with_cancelpt;
#endif
    }
```

patacongo, 2023-07-04:
> The `-ECANCELED` is returned always because of nested cancelation point. The most top `enter_cancellation_point()` is called from `open()`, so all later calls of cancelable APIs from inside the `open()` will be a nested cancelation points:

That is probably the problem.  That cancellation logic was intended to handle the case where calls so cancellation points are nested.  It only supports unwinding back to the first cancellation point where the cancellation can occur.

So I misspoke earlier.  In general, the cancelation does not "latch", but it does latch in the cancellation point logic but only to assure that the call sequence unwinds properly.

That cancellation unwind logic, however, does not work in a loop like the one in nxmutex_lock().  But you have already spotted that.  sem_wait() is not compatible with the logic in nxmutex_lock()!

nxmutex_lock() should call nxsem_wait(), rather that sem_wait() to avoid the nested cancellation point logic.  But that would require a new syscall.  sem_wait() is a system call only because the cancellation point logic needs to run in kernel space.  We would need another nxsem_wait() system call.

I hate to see system calls proliferate.  They should be treating like a precious resource since they add size to the kernel and on many platforms the number of system calls is limited by the instruction set.  So other solutions would be preferable.

patacongo, 2023-07-04:
> The `-ECANCELED` is returned always because of nested cancelation point. The most top `enter_cancellation_point()` is called from `open()`, so all later calls of cancelable APIs from inside the `open()` will be a nested cancelation points:

And this is one of the reasons that we have nx* versions of all most of the application APIs.  Ideally nothing called from open() (or any other application API) should introduce nesting of the cancellation point.  I don't know the full calling sequence but I think suspect that the only violation to this in this case is the expansion of __SEM_WAINT() to sem_wait() instead of to nxsem_wait().

I had proposed at one time that we ASSERT if a nested cancellation point is called.  But I don't think anyone wants to deal with the problems that assertion we be likely to uncover.

pkarashchenko, 2023-07-04:
It is only valid for FLAT build as there is only one version of a libc and there is no way to track if libc API is called by the kernel vs application layer. Some time ago I spent time thinking about a possible solution, but that ended up with nothing.

patacongo, 2023-07-04:
> It is only valid for FLAT build as there is only one version of a libc and there is no way to track if libc API is called by the kernel vs application layer. Some time ago I spent time thinking about a possible solution, but that ended up with nothing.

It seems to me that is not valid in any build configuration.  If the internal nxmutex_lock implementation loops until the return is no -ECANCEL on an nested cancellation point, then an infinite loop will always be generated 100% of the time.  I would not call that "valid" under any configuration.

There is a way ... just call nxsem_wait() instead of sem_wait().  nxsem_wait() is identical EXCEPT that it is not a cancellation point and, hence, will not generate the infinite loop.  You can call nxsem_wait() from nxmutex_lock() in all build configurations (provided that the syscall is made available).

This is a regression that was introduced when the mutex stuff was put in.

pkarashchenko, 2023-07-04:
Yes. It was spotted long time ago and I tried to fix it with https://github.com/apache/nuttx/pull/6376, but those changes didn't work mostly because of `include/mutex.h` that was interfering with third party `mutex.h` and now this bites me back. Let me try to re-evaluate things again. I recall that I was thinking of creating a `_MUTEX_XXX` macro interface and redirect it either to `pthread_mutex_` or `nxmutex_` interface, but need to carefully inspect all the places again

patacongo, 2023-07-04:
> I recall that I was thinking of creating a `_MUTEX_XXX` macro interface and redirect it either to `pthread_mutex_` or `nxmutex_` interface, but need to carefully inspect all the places again

Consider my recommendation to just call nxsem_init() instead of __SEM_INIT().  That should cleanly eliminate the problem as far as I can tell.

pkarashchenko, 2023-07-04:
Since currently `nxmutex_` interface is used in libc it will simply not link in app build in case of `CONFIG_BUILD_KERNEL=y`

patacongo, 2023-07-04:
> Since currently `nxmutex_` interface is used in libc it will simply not link in app build in case of `CONFIG_BUILD_KERNEL=y`.

Nor for CONFIG_BUILD_PROTECTED=y

> Seems like we need a new system call, but that is bad if sys call is done from inside the kernel.

There are no system calls inside the kernel.  They exist only in user space in the PROTECTED and KERNEL builds.

When you build the system calls, two libraries are generated:  (1) a "proxy" library that invokes the system call and (2) a "stub" library that receives the system call.

That is, the system call proxy library is linked only into user space; the system call stub library is linked only into kernel space.  So you can never make system calls from within the kernel.

nxsem_wait() would perform the system call and exist only in user space.  If user space code calls nxsem_wait(), it will invoke a system call.

There is no such system call in the kernel space.  If kernel space code calls nxsem_wait(), the actual function nxsem_wait() will be called.

So nxsem_wait() can be called from indirectly from nxmutex_lock() in libc OR directly from within the kernel.  It all works with no weird behavior.

pkarashchenko, 2023-07-05:
So we need to name `nxsem_wait()` a system call, right?

patacongo, 2023-07-05:
> So we need to name `nxsem_wait()` a system call, right?

Yes, and change the call in nxmutex_lock to nxsem_init (and fix up handling of the return value).  That nxmutex_lock change is all that is needed in FLAT mode so you can test the solution if you like.

I have lots of free time these days since I am not so involved in the project.  So, if you want me to, I can implement the change.

pkarashchenko, 2023-07-05:
Some time ago after reading https://github.com/apache/nuttx/issues/1027 I thought that making NX version of an API to be system calls may be a solution for get rid of all the `#if !defined(CONFIG_BUILD_FLAT) && defined(__KERNEL__)` and fixing `errno` access inside the kernel as well and even started some work in https://github.com/apache/nuttx/pull/6433
Not sure if that activity and this one are related, but seems to me from the first look.

patacongo, 2023-07-06:
> I would appreciate if you can handle the `nxsem_` related part as I'm a bit busy with other work this days.

Okay.  I will look at this a little later today.

pkarashchenko, 2023-07-06:
> @pkarashchenko I just created #9716. Can you describe how you re-create the problem. Do you just need to enable cancellation points an cat a file from NSH? Or is there is particular test that should be run?

I'm having the system that has "manager task" and "worker tasks". "Worker tasks" are spawned based on configuration located on SD card. If new configuration is detected by "manager task" it shut down all current "worker tasks" and spawns new set of "worker tasks". There is no trivial way to reproduce the issue (I do not have ostest doing that, but maybe some can be written). One of "worker tasks" sends CAN messages periodically with `open()` -> `write()` -> `close()` sequence. The issue happens when "manager task" tried to shut down "worker task" sensing CAN message using `task_delete()` -> `waitpid()` sequence. The `CONFIG_CANCELLATION_POINTS` is `y`.

closed 2023-07-06 by acassis

patacongo, 2023-07-06:
It is worth nothing that there is a blatant violation of explict POSIX requirements if mutex_lock() logic is permitted to EVER call sem_wait().  Why?  Because POSIX explicit forbids the creation of any cancellation points other that those called out in the POSIX specifications.

If mutex_lock() calls sem_wait(), then mutex_lock() becomes a cancellation point.  But the POSIX requirement is: "An implementation shall not introduce cancellation points into any other functions specified in this volume of IEEE Std 1003.1-2001." https://pubs.opengroup.org/onlinepubs/009695399/functions/xsh_chap02_09.html

pkarashchenko, 2023-07-06:
Yes. Things are a bit messy now and I'm willing to help in cleaning things up. That is why some times ago I started with an attempt to remove nested cancellation points.y plan was to go API by API from cancellation points list and check for implicit cancellation points.