apache/nuttx #9712: Help wanted for testing and improving real time performance  :)

GooTal, 2023-07-06:
Hi, i learned that NuttX is a RTOS. 

Is there any document describing the design of real time mechanism in detail? I could search the key word \`real time\` from source code but these words seem scattered.

I\`d like to test the real time performance and try to improve it (I\`ll try my best). I\`ve learned from some papers that stress-ng could be the load to the system. Cyclictest and TACLe suits are usually used. Do i need to port these programs? Or are there already some test suits which could measure the real time performance?

Thanks a lot.

patacongo, 2023-07-06:
> Is there any document describing the design of real time mechanism in detail? I could search the key word `real time` from source code but these words seem scattered.
>
> I`d like to test the real time [...]

Nuttx was designed initially to support rate monotonic scheduling (RMS).  The system is relatively easy to add other real time schedulers, but no new real time schedulers have been added to the system.

Rate monotonic scheduling refers primarily to how you prioritize you real time tasks, rather than to a type of scheduler.  In fact it assumes strict priority scheduling (SCHED_FIFO).  If you Google for "Rate Monotonic Scheduling" you can find lots of information about the algorithm so no additional documentation should be required.

The responsibility of the RTOS for RMS are to support the underlying assumptions of RMS.  These _idealized_ assumptions are (https://en.wikipedia.org/wiki/Rate-monotonic_scheduling):

- No resource sharing (processes do not share resources, e.g. a hardware resource, a queue, or any kind of semaphore blocking or non-blocking (busy-waits))
- Deterministic deadlines are exactly equal to periods
- Static priorities (the task with the highest static priority that is runnable immediately preempts all other tasks)
- Static priorities assigned according to the rate monotonic conventions (tasks with shorter periods/deadlines are given higher priorities)
- Context switch times and other thread operations are free and have no impact on the model

These are all ideal and no real time system can meet all of the assumptions.  The biggest risk is the proliferation of locks, mutexes, and semaphores which destroy real time performance.

acassis, 2023-07-06:
Hi @GooTal as Mr. Greg said currently for real time application you can take a look at the Rate Monotonic implemented on NuttX, if you want to implement another scheduler I think EDF is an good candidate:
https://en.wikipedia.org/wiki/Earliest_deadline_first_scheduling

Linux kernel has support to EDF (https://en.wikipedia.org/wiki/SCHED_DEADLINE) since version 3.14.

For realtime performance in general, I think @raiden00pl @robertobucher and @sebastianene07 did some tests and experiments. @sebastianene07 contributed support for realtime analysis using the NuttX SIMulator.

GooTal, 2023-07-10:
Thanks a lot @patacongo @acassis. I\`ll do some research. :)

mu578, 2023-07-10:
@GooTal  seek for Yahia Farghaly, I don't remember his anchor it is from LKin, he did some interesting case of study about various static and dynamic scheduling.