apache/nuttx #9718: which is the last correct compile version on cygwin?

wangfandasha, 2023-07-07:
Hi:
I have try the master branch and remotes/origin/releases/12.2 both get the rusult as following.When I use nuttx 2 years ago,it can run correctly.Is it  a new problem or there is some thing I configure is not correct?
```
Create version.h
LN: platform/board to /cygdrive/d/openSource/nuttx-git/apps/platform/dummy
Register: hello
Register: nsh
Register: sh
./builtin_list.c:25:10: fatal error: nuttx/config.h: No such file or directory
   25 | #include <nuttx/config.h>
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
ERROR: arm-none-eabi-gcc failed: 1
       command: arm-none-eabi-gcc -MT ./builtin_list.c.cygdrive.d.openSource.nuttx-git.apps.builtin.o  -M '-Wstrict-prototypes' '-Os' '-fno-strict-aliasing' '-fomit-frame-pointer' '-mthumb' '-Wa,-mthumb' '-Wa,-mimplicit-it=always' '-fno-common' '-Wall' '-Wshadow' '-Wundef' '-ffunction-sections' '-fdata-sections' '-mlittle-endian' '-march=armv7e-m' '-mtune=cortex-m4' '-mfloat-abi=soft' '-isystem' '/cygdrive/d/openSource/nuttx-git/nuttx/include' '-D__NuttX__' '-DNDEBUG' '-D__KERNEL__' '-pipe' '-I' '/cygdrive/d/openSource/nuttx-git/apps/include' ./builtin_list.c
make[2]: *** [/cygdrive/d/openSource/nuttx-git/apps/Application.mk:257: .depend] Error 1
make[1]: *** [Makefile:57: /cygdrive/d/openSource/nuttx-git/apps/builtin_depend] Error 2
make: *** [tools/Unix.mk:601: pass2dep] Error 2
```

wangfandasha, 2023-07-07:
my system is windows 10 and compile environment is cygwin

acassis, 2023-07-08:
@zouboan did you get this issue on Windows?

@wangfandasha you can also use WSL on Windows 10, it is faster than Cygwin

wangfandasha, 2023-07-10:
> @zouboan did you get this issue on Windows?
> 
> @wangfandasha you can also use WSL on Windows 10, it is faster than Cygwin

OK，now I install vmware and It run well.whenever,It's a little of troublesome.

zouboan, 2023-07-10:
@acassis Sorry,I haven't tried cygwin before

patacongo, 2023-07-10:
In the GCC command line, I see things like:

    '-I' '/cygdrive/d/openSource/nuttx-git/apps/include'

What toolchain are you using?  Is it a Windows native toolchain like ARM Embedded Toolchain?  Or is it a custom toolchain built under Cygwin.

A Windows native toolchain cannot handle a POSIX path like that.  It would want 'D:\openSource\nuttx-git\apps\inclde' instead.

The toolchain selection is controlled in the configuration.  If you are building in Cygwin, you need to select the host:  CONFIG_WINDOWS_CYGWIN . And if you are using a Windows native toolchain, you must select CONFIG_WINDOWS_NATIVE

If you build a GCC toolchain under Cygwin, the Cygwin works exactly like the a Linux build and that should work fine.  The Cygwin build with the Windows native toolchain is much more complex due to all of the path modifications that are required by the tool.

You should be able to build any toolchain under Cygwin if you have the dependencies installed.  One option is bitbucket.org/nuttx/buildroot but that is very old.

patacongo, 2023-07-10:
Might be related to #3799

acassis, 2023-07-10:
> @acassis Sorry,I haven't tried cygwin before

@zouboan are you using Msys / Msys2 ?

BTW, not with CMake I think real native build on Windows could be possible!