apache/nuttx #9738: Logic in mutex_lock() breaks thread cancellation

patacongo, 2023-07-07:
The logic appears in `nxmutex_lock()`:

    195 int nxmutex_lock(FAR mutex_t *mutex)
    196 {
    197   int ret;
    198
    199   DEBUGASSERT(!nxmutex_is_hold(mutex));
    200   for (; ; )
    201     {
    202       /* Take the semaphore (perhaps waiting) */
    203
    204       ret = nxsem_wait(&mutex->sem);
    205       if (ret >= 0)
    206         {
    207           mutex->holder = _SCHED_GETTID();
    208           break;
    209         }
    210       else if (ret != -EINTR && ret != -ECANCELED)
    211         {
    212           break;
    213         }
    214     }
    215
    216   return ret;
    217 }

`mutex_lock()` is not a cancellation point (and cannot be because POSIX forbids the creation of unspecified cancellation points).  As a consequence, `mutex_lock()` may NOT be used by any cancellation point function.  If a cancellation point function calls `mutex_lock()`, then the cancellation event will be lost and thread cancellation will be broken.

For example,` open()` is a cancellation point that calls `mutex_lock()`, but the `open()` cancellation point (as well as others) is broken.  This is because in the event of cancellation, `nxsem_wait()` will return -ECANCEL.  In order for the cancellation to work properly, the -ECANCEL error needs to propagate all the way back to the cancellation point function, `open()` in this case.  Then, the thread exit will occur.

However, if -ECANCEL is received by `nxmutex_lock()`, it will simply loop and call `nxsem_wait()` again. ` nxsem_wait()` will NOT report -ECANCEL again.  That cancellation is not latched and the cancellation event is lost.  So the thread cannot be canceled.

patacongo, 2023-07-07:
Related to #9716.  That PR changes this same section of code, but does not fix the broken thread cancellation

patacongo, 2023-07-07:
It seems like a solution to this regression would require that `nxmutex_lock()` know that it is within a cancellation point and, if so, return -ECANCEL without looping.  This is knowable by testing `tcb->cpcount` which would be set to a non-zero value by previous calls to `enter_cancellation_point()`

pkarashchenko, 2023-07-07:
I think this is not exactly true or true only for nested cancellation points. If for example `open()` is called and cancelation point is entered. Then `nxmutex_lock` is called and it blocks waiting on a semaphore with `nxsem_wait()`. If cancellation is requested then `nxsem_wait()` will resume and `-ECANCEL` will be returned. That will lead to an attempt to call `nxsem_wait()` for a second time because `else if (ret != -EINTR && ret != -ECANCELED)` will be evaluated to `false`, but the second time the `nxsem_wait()` will not return `-ECANCEL` and will continue waiting on a semaphore.
The original issue fixed by https://github.com/apache/nuttx/pull/9716 is because of nested cancellation point occur and `nxmutex_` does not allow this, so switching from `see_wait()` to `nxsem_wait()` resolves it.

pkarashchenko, 2023-07-07:
Even before `nxmutex_` was introduced the VFS used
```
      /* Decrement the references of the inode */

      do
        {
          ret = inode_lock();

          /* This only possible error is due to cancellation of the thread.
           * We need to try again anyway in this case, otherwise the
           * reference count would be wrong.
           */

          DEBUGASSERT(ret == OK || ret == -ECANCELED);
        }
      while (ret < 0);
```
in `inode_release()` that is called from `open()`. The introduction of `nxmutex_` just implicitly propagated similar behaviour to many other places.

pkarashchenko, 2023-07-07:
There obviously is an issue in code, but we need to review all the cancellable APIs one by one and make sure that no implicit cancellation points are introduced to any other API except those that are listed by POSIX

patacongo, 2023-07-07:
> I think this is not exactly true or true only for nested cancellation points. If for example `open()` is called and cancelation point is entered. Then `nxmutex_lock` is called and it blocks waiting [...]

Yes, that is exactly the problem.  The thread cannot be cancelled.  That is what this issue is saying.  Any thread that calls a function loops on -ECANCEL without returning the condition cannot be cancelled.  That is a bug; it is a failure to cancel the thread.

> The original issue fixed by #9716 is because of nested cancellation point occur and `nxmutex_` does not allow this, so switching from `see_wait()` to `nxsem_wait()` resolves it.

No, it does not.  The only difference is that sem_wait() returned -ECANCEL repeatedly which caused the tight, infinite loop.  Calling nxsem_wait() instead fixes that infinit loop.  But now it calls nxsem_wait() just waits on the semaphore again and losses the cancelation condition.  That is a bug; the thread cannot be cancelled.

patacongo, 2023-07-07:
> There obviously is an issue in code, but we need to review all the cancellable APIs one by one and make sure that no implicit cancellation points are introduced to any other API except those that are listed by POSIX

And cancellable APIs cannot call any function that ignores the -ECANCEL condition or it fails to be cancelled.

patacongo, 2023-07-07:
> Even before `nxmutex_` was introduced the VFS used
>
> ```
>       /* Decrement the references of the inode */
>
>       do
>         {
>           ret = inode_lock();
>
>           /* This only possible error is [...]

Yes, but the comment is intended to point out that this logic is wrong.  It breaks cancel-ability, but is necessary ONLY to preserve the reference count.  This is a known error that was never fixed.

pkarashchenko, 2023-07-07:
> Yes, that is exactly the problem. The thread cannot be cancelled. That is what this issue is saying. Any thread that calls a function loops on -ECANCEL without returning the condition cannot be [...]

Thread will be canceled as soon as it reaches the `leave_cancellation_point()` after semaphore will be acquired. That is not an optimal solution I would say, but still I cannot agree that "thread cannot be cancelled", but more like "deferred cancelled".

closed 2023-07-08 by patacongo

mu578, 2023-07-08:
Another option would be to have nxsem_timed_wait call with a ETIMEDOUT condition thus you could restart if interrupted by handler instead waiting forever.