apache/nuttx #9775: Experiencing system crashes with upd sendto

Shark7Pepper, 2023-07-11:
Version nuttx8.2. mtu is set to 1500.When I transfer data using udp protocol, using function sendto() with parameter len set to 4096, the system crashes. I want to know why? Does 8.2 support ip segmentation and how do I configure the system?

acassis, 2023-07-11:
@Shark7Pepper did you try to reproduce the issue in a recent NuttX version? If the issue still happening in the mainline we will fix it.

xiaoxiang781216, 2023-07-11:
it's possible now If the new ip fragment is enabled.:)

fjpanag, 2023-07-11:
@Shark7Pepper can you please describe the "crash" that you are facing?

This has been discussed [here](https://lists.apache.org/thread/rlf40pkcqdxn15h6gtzl50m6fz0f5xt1), and it was fixed in #9423. Is this the same issue that you are facing?

Which arch are you using? At least in the STM32 driver there seems to be an issue that can also cause a crash with large datagrams.