apache/nuttx #9840: Experiencing possible race condition with pthread_cond_timedwait.

g2gps, 2023-07-17:
I'm using a combination conditional variables and mutex's for thread synchronization on a shared resource. I seem to be experiencing a race condition, in which a signal on a condition variable is missed, which results in the application logic becoming locked, with every thread is waiting on my interlocking mutex's.

I've attempted to extract the problematic logic out of our existing application, into a minimal reproducible example, which still encounters the issue:

```c
#include <assert.h>
#include <err.h>
#include <fcntl.h>
#include <pthread.h>
#include <stdbool.h>
#include <stdio.h>
#include <stdlib.h>
#include <sys/time.h>
#include <unistd.h>

#include <debug.h>

// clang-format off
struct trigger_cmd {
  pthread_mutex_t mutex;
  pthread_cond_t cmd_cond;
  pthread_cond_t result_cond;
  enum Type {
    trg_NONE,
    trg_START,
    trg_HALT,
  } trigger;
} core = {
    PTHREAD_MUTEX_INITIALIZER, 
    PTHREAD_COND_INITIALIZER,
    PTHREAD_COND_INITIALIZER,
    trg_NONE
};
// clang-format on

pthread_mutex_t serialise = PTHREAD_MUTEX_INITIALIZER;

size_t get_random(size_t min, size_t max) {
  FILE *f = fopen("/dev/urandom", "rb");
  if (f == NULL) {
    printf("Can't get random number\n");
    exit(EXIT_FAILURE);
  }

  size_t buffer;
  int count = fread(&buffer, sizeof(size_t), 1, f);
  fclose(f);
  if (!count) {
    printf("Can't get random number\n");
    exit(EXIT_FAILURE);
  }
  return min + (buffer % max);
}

void busyloop() {
  size_t us = get_random(0, 1000000);
  sinfo("Sleep for %u\n", us);
  usleep(us);
  sinfo("Wake\n");
}

void trigger_start() {
  sinfo("Lock serialise: core %d, ser %d\n", core.mutex.pid, serialise.pid);
  pthread_mutex_lock(&serialise);
  sinfo("Lock core: core %d, ser %d\n", core.mutex.pid, serialise.pid);
  pthread_mutex_lock(&core.mutex);

  core.trigger = trg_START;

  sinfo("Signal on cmd: core %d, ser %d\n", core.mutex.pid, serialise.pid);
[... 261 lines cut ...]

      goto out;
    }
```

From `sched/pthread/pthread_condclockwait.c`:
```c

      sinfo("Give up mutex...\n");

      /* We must disable pre-emption and interrupts here so that
       * the time stays valid until the wait begins.   This adds
       * complexity because we assure that interrupts and
       * pre-emption are re-enabled correctly.
       */

      sched_lock();
      flags = enter_critical_section();
```

If that is the case, I could see how the absolute time given to pthread_cond_timedwait may have expired before `nxsem_wait` is called in `nxsem_clockwait`. However, I would have through that the scheduler is locked between when the mutex is released and the wait is started, so this shouldn't have any effect.

I've spent a bit of time trying to understand the interactions, but I don't believe I have a full understanding of the implementation in Nuttx to make any feasible progress. At this point I'm not sure if it's an issue with the `Litex` port, the kernel configuration, or an oversight in our application. 

I'd appreciate any assistance tracking down the issue I'm seeing.

acassis, 2023-07-25:
@g2gps did you try it on some ARCH more stable like STM32 or ESP32?

pkarashchenko, 2023-07-25:
Also it would be good if you can try if disabling priority inheritance gives any change to reproduction rate.

g2gps, 2023-08-16:
Thanks for the feedback. Enabling or disabling priority inheritance doesn't have a noticeable effect on the time until failure.

I did try building and running on an `esp32-devkitc` and the following configuration.
```
# CONFIG_ARCH_LEDS is not set
# CONFIG_NSH_ARGCAT is not set
# CONFIG_NSH_CMDOPT_HEXDUMP is not set
CONFIG_ARCH="xtensa"
CONFIG_ARCH_BOARD="esp32-devkitc"
CONFIG_ARCH_BOARD_COMMON=y
CONFIG_ARCH_BOARD_ESP32_DEVKITC=y
CONFIG_ARCH_CHIP="esp32"
CONFIG_ARCH_CHIP_ESP32=y
CONFIG_ARCH_CHIP_ESP32WROVER=y
CONFIG_ARCH_STACKDUMP=y
CONFIG_ARCH_XTENSA=y
CONFIG_BOARD_LOOPSPERMSEC=16717
CONFIG_BUILTIN=y
CONFIG_DEBUG_ASSERTIONS=y
CONFIG_DEBUG_ASSERTIONS_EXPRESSION=y
CONFIG_DEBUG_ERROR=y
CONFIG_DEBUG_FEATURES=y
CONFIG_DEBUG_INFO=y
CONFIG_DEBUG_SCHED=y
CONFIG_DEBUG_SCHED_ERROR=y
CONFIG_DEBUG_SCHED_INFO=y
CONFIG_DEBUG_SCHED_WARN=y
CONFIG_DEBUG_WARN=y
CONFIG_DEV_URANDOM=y
CONFIG_ESP32_UART0=y
CONFIG_FS_PROCFS=y
CONFIG_IDLETHREAD_STACKSIZE=3072
CONFIG_INIT_ENTRYPOINT="nsh_main"
CONFIG_INTELHEX_BINARY=y
CONFIG_LIBM=y
CONFIG_MM_REGIONS=3
CONFIG_NSH_ARCHINIT=y
CONFIG_NSH_BUILTIN_APPS=y
CONFIG_NSH_FILEIOSIZE=512
CONFIG_NSH_LINELEN=64
CONFIG_NSH_READLINE=y
CONFIG_PREALLOC_TIMERS=4
CONFIG_RAM_SIZE=114688
CONFIG_RAM_START=0x20000000
CONFIG_RR_INTERVAL=200
CONFIG_SCHED_WAITPID=y
CONFIG_START_DAY=6
CONFIG_START_MONTH=12
CONFIG_START_YEAR=2011
CONFIG_SYSLOG_BUFFER=y
CONFIG_SYSLOG_PROCESSID=y
CONFIG_SYSLOG_PROCESS_NAME=y
CONFIG_SYSTEM_NSH=y
CONFIG_UART0_SERIAL_CONSOLE=y
```

Same result. After a period of time, the signal is missed.

g2gps, 2024-05-15:
This is an issue with our application logic and not NuttX.

What's happening is the  `cmd_cond`  conditional variable is being signaled after the abs time in `pthread_cond_timedwait` has expired, and before the mutex is relocked this function returns. From the documentation on [opengroup](https://pubs.opengroup.org/onlinepubs/9699919799/), this is the expected behavior of this interface.

closed 2024-05-15 by g2gps