apache/nuttx #9845: xtensa/risc-v issue with module app (after PR #7202 and #9737)

tmedicci, 2023-07-17:
Although it's related to an app, this issue has been raised after PRs #7202 and #9737.

Both `nuttx/boards/risc-v/esp32c3/esp32c3-devkit/configs/module/defconfig` and `nuttx/boards/xtensa/esp32/esp32-devkitc/configs/module/defconfig` build the `module` app example and fails to run it.

Considering ESP32-C3, before https://github.com/apache/nuttx/commit/fec789047a (from #7202), we had:
```
nsh> module
main: Registering romdisk at /dev/ram0
module_initialize:
chardev_read: Returning 36 bytes
chardev_read: Returning (0x3fc85aa0):
0000  48 69 20 74 68 65 72 65 2c 20 61 70 70 73 2f 65  Hi there, apps/e
0010  78 61 6d 70 6c 65 73 2f 6d 6f 64 75 6c 65 20 74  xamples/module t
0020  65 73 74 0a                                      est.            
main: Bytes read (0x3fc85aa0):
0000  48 69 20 74 68 65 72 65 2c 20 61 70 70 73 2f 65  Hi there, apps/e
0010  78 61 6d 70 6c 65 73 2f 6d 6f 64 75 6c 65 20 74  xamples/module t
0020  65 73 74 0a                                      est.            
chardev_write: Writing 26 bytes
chardev_write: Writing (0x3c0272ac):
0000  48 69 20 74 68 65 72 65 20 69 6e 73 74 61 6c 6c  Hi there install
0010  65 64 20 64 72 69 76 65 72 0a                    ed driver.      
main: Bytes written (0x3c0272ac):
0000  48 69 20 74 68 65 72 65 20 69 6e 73 74 61 6c 6c  Hi there install
0010  65 64 20 64 72 69 76 65 72 0a                    ed driver.      
module_uninitialize: arg=0
main: Mounting ROMFS filesystem at target=/mnt/romfs with source=/dev/ram0
main: Read 36 bytes
main: Wrote 26 bytes
nsh>
```

Just after https://github.com/apache/nuttx/commit/fec789047a (from #7202), we have the following errors:
```
nsh> module
main: Registering romdisk at /dev/ram0
main: Mounting ROMFS filesystem at target=/mnt/romfs with source=/dev/ram0
ERROR: insmod(/mnt/romfs/chardev, chardev) failed: 22
nsh>
```

And after https://github.com/apache/nuttx/commit/b5cd7c2a82 (from #9737) , we got stuck at:
```
nsh> module
main: Registering romdisk at /dev/ram0
main: Mounting ROMFS filesystem at target=/mnt/romfs with source=/dev/ram0

```

tmedicci, 2023-07-19:
@nealef , do you have any clue about this issue? It seems to be something related to the archs other than arm...

nealef, 2023-07-19:
I have a fix that affects this in test at the moment.

tmedicci, 2023-07-19:
> I have a fix that affects this in test at the moment.

Perfect, let me know if you need anything about it (I can run tests too). Thanks!

acassis, 2023-07-26:
@tmedicci could you confirm that issue was fixed and then please close this issue

tmedicci, 2023-07-27:
Yes, it has been fixed properly. Checked with internal CI ;)

closed 2023-07-27 by tmedicci