apache/nuttx #9850: macOS related build failure

mks2183, 2023-07-18:
Hi All,

I am seeing error in build where PR #9846. Is it something related with macOS ? Did anybody face it ?

Regards,
Manish

acassis, 2023-07-18:
@mks2183 I think the issue is related to some MacOS downloaded file. I don't know much about the CI, but maybe we need to create a dedicated mirror to avoid these frequent errors on MacOS.

btashton, 2023-07-18:
> @mks2183 I think the issue is related to some MacOS downloaded file. I don't know much about the CI, but maybe we need to create a dedicated mirror to avoid these frequent errors on MacOS.

As I have said before this error has nothing to do with downloading a file. Why do you keep saying that.  There is no world where we should be mirroring pypi.

acassis, 2023-07-18:
You are right @btashton, this is not related to download timeout or similar as in the past, seems like some modification in the pypi files itself. But having a controlled mirror we could make the packages fixed or only update when after testing some staging CI before updating to use in the main CI.

mu578, 2023-07-18:
This a cython 3.0 issue, it is broken within python 3.12 ; till today the only solution that is to get prev version install, meanwhile, I agree relaying on pip is tricky, I have that experience to bind a lot of numerical / ML /algebra algorithms from C to python using cython, I banned that package manager long ago, I do maintain my own toolchain separated, why? because it randomly breaks packages.

xiaoxiang781216, 2023-07-18:
Anyway, the ci is broken now. If nobody can fix it, I will disable macOS ci tomorrow.

xiaoxiang781216, 2023-07-19:
Fixed by https://github.com/apache/nuttx/pull/9855.

closed 2023-07-19 by xiaoxiang781216

mu578, 2023-08-02:
https://github.com/apache/nuttx/pull/9993#issuecomment-1662509586