apache/nuttx #9875: Allow 100 column lines

mks2183, 2023-07-23:
Hi All,

Code readability is compromised because code cannot go beyond 80 columns.  I propose to allow the 80 columns as standard for ease of coding. Shifting to 100 columns will get rid-off unnecessary multiline coding, unnecessary whitespace changes in files which may decrease readability in long run.

Regards,
Manish

patacongo, 2023-07-23:
Lines are not limited to 80 characters in any way.  They are limited to match the

    /**********....********/

comments.  These serve as "rulers".  Lines may not exceed the length of those rulers.

If you need longer lines for some reason, you use longer rulers.  We won't change the standard because there is no such standard.

patacongo, 2023-07-23:
I do almost everything on a laptop with a 14 inch screen.  I is nearly impossible to use a laptop with wide files (unless you want to deal with a microscopic font or weird horizontal scrolling).

mu578, 2023-07-23:
this day, myself I rule mine at 110, I can't anymore work on too small screens, even when I do things on smartphones I use a looking glass, too tiny, can't work at less than 17″ anymore, kills my eyes, hopefully, I didn't end up loosing vision, but at some point, I was feeling that if I continued the constant abuse I'd get irreversible issues, so I bailed out in time, wearing protective glasses too.

patacongo, 2023-07-23:
> These days, I rule mine at 110, I can't anymore work on too small screens, even when I do things on smartphones I use a looking glass, too tiny, can't work at less than 17″ anymore, kills my eyes,

A 17 inch laptop is not very portable.  For most things, 14 is pretty ideal... except for wide files.  80 characters is not too bad.  Using a magnifier is not a good solution because it is hard to see the overall flow of the files.

I think the way that things work now with the "ruler" is as close to ideal as you can get.  It allows a lot of flexibility but still has a consistent look and feel.  There are some files with line widths of 128 or more.  That is especially common with register definition header files where you would like to have a full definition on one line

mks2183, 2023-07-24:
Respected Sir @patacongo @mu578,

You are correct and I do agree on your things. However, I would like to present few things on your table before we conclude this discussion:

- With my recent talks to my colleagues, they are now using bigger laptop so 14' inch is very less in trend
- I've observed in my organization and other organizations too, they are bigger screen size laptops/desktop. 
- I've checkout in Linux forum that people are using 100 lines as a non-strict policy.
- Multiline are always not good otherwise you have to make font size small which is not good for human eyes.

I would like to propose to have a non-strict 100 lines as a width for better screen to eye ratio which will definitely help novices like me and others to have better code readability and for better eyesight.  

Sincere Regards,
Manish Sharma

TimJTi, 2023-07-24:
Not 100% accurate. Style guide requires 78 characters unless necessary:“Line Width. Text should not extend past column 78 in the typical C source or header file. Sometimes the nature of the content of a file may require that the lines exceed this limit. This often occurs in header files with naturally long definitions. If the line width must extend 78 lines, then some wider line width may be used in the file provided that it is used consistently.”I have stuck to 78 to avoid PRs being rejected and haven’t found it any issu to be honest.

mks2183, 2023-07-24:
Thanks sir @TimJTi. 

Regards,
Manish

closed 2023-07-24 by mks2183

mu578, 2023-07-26:
yes, it is a policy, but it has been inherited from past and gone limitations for few decades now, I suggest grown ups to let pass to the next generation (that's true maturity to know to let it go), BTW that's all the mindset issue we are confronted regarding C standard and the inability to actually standardize any on a long run.

It reminded me this silly argument about entabbing or not, we were not entabbing because 30 years ago editors were not properly handling tab control character, that's it, use tabs by any means the argument has been settled 25 years ago. And I hear those delusional arguments on it still going... jeez, they don't even no the basic rationality about it.