apache/nuttx #9876: Configuration still points to the older board used as template

tcpaiva, 2023-07-23:
Hi, all. 

I don't think this is a bug. I am probably missing some very obvious step.

I've been trying to port Nuttx to a new platform using IMX6 (`sabre-6quad`) as a start point since it also uses `armv7-a`.

After copying and changing all the required files with respect to  `arch/arm/include/"mychip"`, `arch/arm/src/"mychip"`, `boards/arm/"mychip"/"myboard", including `boards/Kconfig`, when I run `./tools/configure.sh "myboard":nsh`, it still uses the IMX6 paths. 

I didn't change the file `boards/arm/"mychip"/"myboard"/configs/nsg/defconfig`, since it says that it would be created automatically.

Thanks for the help and keep it up! Sorry if any information is missing. Please let me know and I will complete this description.

PS: I am using GNU/Linux as the operating system.

acassis, 2023-07-23:
Hi @tcpaiva nice to hear from you!

Before running "./tools/configure.sh \"myboard\":nsh" you need to manually modify boards/arm/"mychip"/"myboard"/configs/nsh/defconfig to replace any references from IMX6, SABRE-6QUAD, etc to your board. Pay special attention in this process because any "CONFIG_" name incorrect will generate strange errors later. Also the board directory name needs to be correct.

You just need to do it once, after that every time you modify the menuconfig and run "make savedefconfig" it will create a sane defconfig to be used for new config. You don't need (read: shouldn't) to modify manually later.

This is a chicken-egg dilemma, to create a correct defconfig you need an initial defconfig configured for your board. AFAIK there is not a tool to create the initial "egg" for you.

tcpaiva, 2023-07-24:
Hi, @acassis! Good to hear from you as well! You are very active here as always. Thanks for the prompt feedback. :grin: 

I tried what you mentioned and when I run `./tools/configure.sh -l <myboard>:nsh`, I get:

```
$ ./tools/configure.sh -l <myboard>:nsh
Sun Jul 23 08:51:27 PM EDT 2023
  Copy files
  Select CONFIG_HOST_LINUX=y
  Refreshing...
CP: arch/dummy/Kconfig to /path/to/nuttx/arch/dummy/dummy_kconfig
CP: boards/dummy/Kconfig to /path/to/nuttx/boards/dummy/dummy_kconfig
LN: platform/board to /path/to/apps/platform/dummy
LN: include/arch to arch/arm/include
LN: include/arch/board to /path/to/nuttx/boards/arm/<mychip>/<myboard>/include
LN: drivers/platform to /path/to/nuttx/drivers/dummy
LN: include/arch/chip to /path/to/nuttx/arch/arm/include/<mychip>
LN: arch/arm/src/chip to /path/to/nuttx/arch/arm/src/<mychip>
LN: arch/arm/src/board to /path/to/nuttx/boards/arm/<mychip>/<myboard>/src
(...)
```

However, if I execute `make menuconfig` just after, I get:
```
$ make menuconfig                     
Sun Jul 23 08:54:29 PM EDT 2023
CP: arch/dummy/Kconfig to /path/to/nuttx/arch/dummy/dummy_kconfig
CP: boards/dummy/Kconfig to /path/to/nuttx//Kconfig
LN: platform/board to /path/to/apps/platform/dummy
LN: include/arch to arch/arm/include
LN: include/arch/board to /path/to/nuttx//include
LN: drivers/platform to /path/to/nuttx/drivers/dummy
LN: include/arch/chip to /path/to/nuttx/arch/arm/include/stm32
LN: arch/arm/src/chip to /path/to/nuttx/arch/arm/src/stm32
LN: arch/arm/src/board to /path/to/nuttx//src
No directory at /path/to/nuttx//src
make[1]: *** [tools/Unix.mk:300: arch/arm/src/board] Error 1
make: *** [tools/Unix.mk:674: menuconfig] Error 2
```
I have no idea from where the `stm32` comes from. It seems there is something really strange with my setup. :sweat_smile:

acassis, 2023-07-24:
Normally this error means that your new architecture or board definition is incorrect or missing some symbol definition. 
Note this double slash ("//"), it is the clue you need to config that the symbols were resolved.

For ESP32C3-Devkit board I have this definition:

```
LN: platform/board to /nuttxspace/apps/platform/dummy
LN: include/arch to arch/risc-v/include
LN: include/arch/board to /nuttxspace/nuttx/boards/risc-v/esp32c3/esp32c3-devkit/include
LN: drivers/platform to /nuttxspace/nuttx/drivers/dummy
LN: include/arch/chip to /nuttxspace/nuttx/arch/risc-v/include/esp32c3
LN: arch/risc-v/src/chip to /nuttxspace/nuttx/arch/risc-v/src/esp32c3
LN: arch/risc-v/src/board to /nuttxspace/nuttx/boards/risc-v/esp32c3/esp32c3-devkit/../common
LN: arch/risc-v/src/board/board to /nuttxspace/nuttx/boards/risc-v/esp32c3/esp32c3-devkit/src
```

Probably you have errors in both, in your arch definition and your board definition. Before to create a new arch, I suggest you to just create an iMX6 board, just for test, to confirm your system is sane and to get started with the process. For example you can create an Wandboard i.MX6, you will basically copy the Sabre board and rename the symbols.

tcpaiva, 2023-07-25:
Thanks for the tips, @acassis. I was able to figure out what was wrong in my `Kconfig` files. Now I have `menuconfig` working and I am able to compile the code. It is time to start debugging. =)

closed 2023-07-25 by tcpaiva