apache/nuttx #9885: ESP32 / RMT / WS2812

cederom, 2023-07-25:
The initial code for `RMT` peripheral for `ESP32` has been created by @vbenso in https://github.com/apache/nuttx/pull/6992#issuecomment-1553624904 (AVE!) :-)

Still some updates seems required to make `RMT` work out of the box with `WS2812` devices / examples.

Because there may be more PR required, this is a dedicated Issue to track progress.

cederom, 2023-07-25:
I cannot see the `RMT` peripheral in the `menuconfig` although I can find it througs a search. Could you please check if you can see that peripheral @acassis please ? :-)

[image]
[image]

By the way I noticed that `kconfig` does not print its version anywhere and has to option switch to print version..? I am using `kconfig-frontends-4.11.0.1` on FreeBSD.

vbenso, 2023-07-25:
> I cannot see the `RMT` peripheral in the `menuconfig` although I can find it througs a search. Could you please check if you can see that peripheral @acassis please ? :-)

I think you need to enable the experimental features(or drivers, don't recall the actual wording)

> 
> [image]
> [image]
> 
> By the way I noticed that `kconfig` does not print its version anywhere and has to option switch to print version..? I am using `kconfig-frontends-4.11.0.1` on FreeBSD.
>

cederom, 2023-07-25:
> I think you need to enable the experimental features(or drivers, don't recall the actual wording)

That did the trick, thanks @vbenso :-)

acassis, 2023-07-25:
@cederom did you get it working? Sorry I didn't see you msg before, I disabled GitHub notification because it was too noise

vbenso, 2023-07-26:
Gentleman,

I've submitted the PR to the nuttx-apps branch [https://github.com/apache/nuttx-apps/pull/1849](https://github.com/apache/nuttx-apps/pull/1849).

I hope it helps.
@acassis @cederom 

And please, post some links for videos of your results.

PS.: @cederom I do remember that you I've said something about using inverted logic on the signal line. In that case, I think you can just swap T0H <-> T0L and T1H<->T1L plus changing the RMT_IDLE_OUT_LV_CH0 bit in the RMT_CH0CONF1_REG (reg1_addr in rmt_open)

cederom, 2023-07-26:
Not yet sorry @acassis crazy end of the month.. will report back at next steps :-) Thanks for the dedicated demo @vbenso !! :-)

acassis, 2023-07-26:
@cederom please test using https://github.com/apache/nuttx-apps/pull/1849 and if it worked for you, please close this issue. I bought a 8x32 WS2812 LED matrix on Aliexpress, when it arrive here I will test too

acassis, 2023-07-30:
Everything is in place now, closing the issue!

closed 2023-07-30 by acassis