apache/nuttx #9991: how to use systemview

laoniaokkk, 2023-08-01:
I tried to use sysytemview to reference the sim/segger configuration, but I didn't succeed, and the issue on systemview was also unsuccessful. I would like to know how to go about setting it up to use correctly.

acassis, 2023-08-01:
@xiaoxiang781216 I think we need a concise Documentation like we have to Debug Trace

laoniaokkk, 2023-08-02:
> Please reference this link: #5843

I checked this issue but my issue is not solved, my NUTTX system version is 12.0.0 and my chip is stm32f429. I made sure the program executed up_perf_init() and opened DWT inside. When I click "Start Recording", SystemView doesn't show anything, then click "Stop Recording" to display as below. Can you give me your profile?

[image]

xiaoxiang781216, 2023-08-02:
@Gary-Hobson could you share some defconfig?

raiden00pl, 2023-08-02:
@laoniaokkk you can try this PR https://github.com/apache/nuttx/pull/10006 
There is an example systemview configuration for nucleo-f446re.
The main problem I encountered with enabling systemview is `CONFIG_DRIVERS_NOTERAM=y` which is the default option and at the same time `CONFIG_DRIVERS_NOTE_MAX=1`. This cause systemview support to fail and no error is reported to the user.

laoniaokkk, 2023-08-03:
> @laoniaokkk you can try this PR #10006 There is an example systemview configuration for nucleo-f446re. The main problem I encountered with enabling systemview is `CONFIG_DRIVERS_NOTERAM=y` which is the default option and at the same time `CONFIG_DRIVERS_NOTE_MAX=1`. This causes systemview support to fail and no error is reported to the user.

Yes,I made `CONFIG_DRIVERS_NOTE_MAX=2`, and then the SystemView was work well

closed 2023-08-03 by laoniaokkk