apache/nuttx #10159: Wait after vfork get errno ECHILD

yangyalei, 2023-08-10:
Execute the following code， wait fail with  ret == -1 and errno == ECHILD.
```
    int ret;
    int pid = vfork();
    if (pid == 0)
    {
        execlp("sh", "sh", "-c", "ls", NULL);
    }
    else if (pid > 0)
    {
        int ret = wait(NULL);
        if (ret == -1)
         printf("wait fail from Parent! ret:%d, errno:%d\n", ret, errno);
    }
```

Is it an undefined behavior wait after vfork？But This code test ok on my ubuntu 20.04.
I have lookup vfork and wait defines in some documents, no clear instructions found about this.

So, Do we need to fix this？

xiaoxiang781216, 2023-08-10:
@patacongo @pkarashchenko what your opinion?

patacongo, 2023-08-10:
The logic seems bad.  wait() just calls waitpid() with an invalid PID:

     67 pid_t wait(FAR int *stat_loc)
     68 {
     69   /* wait() is a cancellation point, but nothings needs to be done for this
     70    * trivial case.
     71    */
     72
     73   return waitpid(INVALID_PROCESS_ID, stat_loc, 0);
     74 }

But waitpid() return -ECHILD if the pid is not valid:

    114   ctcb = nxsched_get_tcb(pid);
    115   if (ctcb == NULL)
    116     {
    117       ret = -ECHILD;
    118       goto errout;
    119     }

This logic appears in two places and there are several other cases where -ECHILD is returned.  But if the PID is invalid, I don't think you can get to those other cases.  Has this logic changed recently?

patacongo, 2023-08-10:
> int ret = wait(NULL);

A good test would be to simply replace this with:

    int ret = waitpid(pid);

patacongo, 2023-08-10:
I think I am beginning to understand the real issue here.  My ramblings above are probably not so relevant.  If CONFIG_SCHED_HAVE_PARENT is not defined, then wait() cannot be supported.  Is that correct?  If that is true, then we only need to correct the dependencies to disable wait() if CONFIG_SCHED_HAVE_PARENT is not defined.

yangyalei, 2023-08-11:
Sorry for not explaining my question clearly.

I have enabled CONFIG_SCHED_HAVE_PARENT in my test.
The real question is:
First, in nuttx implement, vfork use waitpid to hang on the parent process, make it continue after child process ended.
https://github.com/apache/nuttx/blob/d98bfc3e49b55ad560907241cbd7e184cf7aac0a/libs/libc/unistd/lib_vfork.c#L66C38-L66C38

Second, this waitpid will make sure child process ended and remove the child process info in nxsched_waitpid.
https://github.com/apache/nuttx/blob/d98bfc3e49b55ad560907241cbd7e184cf7aac0a/sched/sched/sched_waitpid.c#L361C8-L361C62

In the end, if application call wait will find NO cheild process info in 
https://github.com/apache/nuttx/blob/d98bfc3e49b55ad560907241cbd7e184cf7aac0a/sched/sched/sched_waitpid.c#L264

In one world, it looks like vfork interface implement not support to wait vforked child process done, because we don't keep the vforked process info when before it end.

yangyalei, 2023-08-11:
I have submit a PR to this question, So please judge whether we need it.
https://github.com/apache/nuttx/pull/10169

closed 2023-08-11 by patacongo