apache/nuttx #10191: Inconsistent with the expected result when calling cmake module

xuxin930, 2023-08-11:
https://github.com/apache/nuttx/blob/bc181f3f9c98ac527df84cdf0af449c6454f3367/cmake/nuttx_add_library.cmake#L64-L80

I'm migrating some apps libraries to cmake build system. when calling the `nuttx_add_user_library` module function , like this
```cmake
  # In lib-need-to-archive-in-apps/CMakeLists.txt
  nuttx_add_user_library(lib-need-to-archive-in-apps)

  target_sources(lib-need-to-archive-in-apps ${SRCS} )
  target_compile_options(lib-need-to-archive-in-apps ${FLAGS})
  target_include_directories(lib-need-to-archive-in-apps ${INCDIR})
``` 
what I expect is that all output files are archived in libapps, as the description by this function.
But it turns out that it is neither archived in libapps nor involved in the linking of nuttx.

I guess it may be caused by the inheritance of `INTERFACE`, it is not correctly assigned to target apps.
should we change `INTERFACE` in line 76 and line 79 to  `PRIVATE` ？

anchao, 2023-08-11:
Yes, seems make sense, `INTERFACE` just appended to the link interface and are not used for linking to target, you can change it to PRIVATE if you want to use this interface to link with libapps

xuxin930, 2023-08-12:
> Yes, seems make sense, `INTERFACE` just appended to the link interface and are not used for linking to target, you can change it to PRIVATE if you want to use this interface to link with libapps

thanks a lot🙏 , I'm planning  to fix the `nuttx_add_user_library ` function to the expected behavior, which is  archived to libapps.
and also planning to enhance  those **independent static libraries**  that generated by `nuttx_add_libray(foo STATIC)`, make those static libraries have a unified export path target.

closed 2023-08-14 by raiden00pl