apache/nuttx #10197: binfmt need support ST_NOSUID

xiaoxiang781216, 2023-08-11:
as mentioned in https://github.com/apache/nuttx/pull/10176.

acassis, 2023-08-12:
@xiaoxiang781216 there is a discussion about it here: https://stackoverflow.com/questions/27658167/posix-statvfs-required-behaviour

I think POSIX doesn't require it because as pointed there many FS will not support it.

patacongo, 2023-08-14:
ST_NOSUID is a status bit value reported by statvfs() (https://pubs.opengroup.org/onlinepubs/9699919799/functions/fstatvfs.html).  The bit field of reported status bits is required to provided by the f_flag field of the statvfs structure (https://pubs.opengroup.org/onlinepubs/009695399/basedefs/sys/statvfs.h.html).  The only required fields are ST_RDONLY and ST_NOSUID.

These are unrelated to the file system type.  These are specified when the file system is mounted.  The behavior does depend on what is returned by the stat() of the file.  All file systems support ST_NOSUID is some fashion.  If the file system stat() never returns the S_ISUID or S_ISGID flags, then the ST_NOSUID mount flag can have no effect.  If the file system stat() never returns st_uid or st_gid, then the switch to EIUD:EGID of root:root (0:0) is still possible.

I suspect that support for ST_NOSUID flag in binfmt would be independent of the file system type.  If some file systems do not fully support all behaviors, that does not invalidate the requirement.

The mount() interface is, apparently, not specified by POSIX, is left to be implementation defined.  See https://pubs.opengroup.org/onlinepubs/9699919799/xrat/V4_xbd_chap03.html which says:

> Mount Point
>
> The directory on which a "mounted file system" is mounted. This term, like mount() and umount(), was not included because it was implementation-defined.

The NuttX mount() interface generally follows the Linux mount(https://man7.org/linux/man-pages/man2/mount.2.html) semantics. There are numerous Linux mount() flags including the MS_RDONLY and MS_NOSUID flags of interest.  These settings would need to be retained in the i-node of the mount point.  The state of these mount flags would be reflected by the statvfs() ST_RDONLY and ST_NOSUID status bits.

The FreeBSD() mount() interface (https://man.freebsd.org/cgi/man.cgi?query=mount&sektion=2&n=1) is similar and support many mount flags including the flags of interest with similar but different names, MNT_RDONLY and MNT_NOSUID.

patacongo, 2023-08-14:
> If the file system stat() never returns the S_ISUID or S_ISGID flags, then the ST_NOSUID mount flag can have no effect. If the file system stat() never returns st_uid or st_gid, then the switch to EIUD:EGID of root:root (0:0) is still possible.

CLARIFICATION:  No files created on NuttX will ever have S_ISUID or S_ISGID set because the mode bits are ignored by the file creation logic in fs/vfs..  That is another POSIX non-compliance.

The mode bits and the values of UID and GID can be set in file systems like ROMFS where the file system is created using the genromfs tool.  Or also by ad hoc logic such as that which has recently been added for binfs, built=in apps, and ELF modules.

Support is largeky missing and sparse where implemented.  That is why this Issue is important!

xiaoxiang781216, 2023-08-15:
@fxysunshine could you send a PR fix the problem?

patacongo, 2023-08-15:
> If the file system stat() never returns the S_ISUID or S_ISGID mode flags, then the ST_NOSUID mount flag is just ignored. If the file system stat() returns only zero for st_uid or st_gid, then only the switch to EIUD:EGID of root:root (0:0) is still possible.

The creation mode, uid, and gid are already retained in the struct inode., at least for pseudo file system.  It seems like this could be used to create a file-system-independent, common solution.