apache/nuttx #10202: STM32H7 dual core support

raiden00pl, 2023-08-12:
I started working on support for Cortex-M4 core (CPU2) in STM32H7. 
I'm opening this issue to track progress and not duplicate work if someone else is working on it.

My target board for now is nucleo-h745zi, later I will probably focus on arduino boards (portenta/opta).

I'll be doing this in my spare time so I don't have any deadline. If anyone is interested, please feel free to help :)

A loose action plan:

- [ ] initial port for nucleo-h745zi with Cortex-M7 supported [done, not yet upstream]
- [ ] customize linker script to support CPU2 [done, not yet upstream]
- [ ] modify stm32h7 Kconfig to support Cortex-M4 build [done, not yet upstream]
- [ ] fixes to stm32h7x5 port that take into account the presence of CPU2 [started]
- [ ] customize stm32h7 initialization logic for dual core operations (__start, PWR, RCC) [started]
- [ ] `HSEM` driver [started]
- [ ] boot Cortex-M4 core [not working yet]
- [ ] rputn driver based on `HSEM`
- [ ] NSH on Cortex-M4 core available through `cu` on Cortex-M7 core
- [ ] ultimate goal: FOC logic CM4, communication and user interface on CM7

xiaoxiang781216, 2023-08-12:
@raiden00pl so the porting will be an AMP, not SMP.

raiden00pl, 2023-08-12:
I don't know if any RTOS supports STM32H7 as SMP. As far as I know, ST always referred to stm32h7 as AMP in their marketing materials: CM4 for real time work (control algo, network stack, etc) and CM7 for other things.
From my side SMP in this case seems pointless (but interesting as an experiment to see if it's even possible).

acassis, 2023-08-12:
In fact having SMP using different architectures should be an interesting test, since the S of SMP imply Symetric, ARM Cortex-M7 and Cortex-M4 has many similarities. Hey @masayuki2009 have you tested it?

acassis, 2023-08-12:
@raiden00pl did you get OpenOCD working to debug the M4 core?

raiden00pl, 2023-08-12:
@acassis it partly works, breakpoints seem to work but continue after breakpoint and code stepping doesn't. I haven't looked for a reason yet. If there is something wrong with openocd, I'll switch to JLINK hoping it will work :)

masayuki2009, 2023-08-13:
>In fact having SMP using different architectures should be an interesting test, since the S of SMP imply Symetric, ARM Cortex-M7 and Cortex-M4 has many similarities. Hey @masayuki2009 have you tested it?

Since the Cortex-M7 has D-cache, it would be difficult to work with Cotex-M4 in SMP mode.

acassis, 2023-08-13:
> > In fact having SMP using different architectures should be an interesting test, since the S of SMP imply Symetric, ARM Cortex-M7 and Cortex-M4 has many similarities. Hey @masayuki2009 have you tested it?
> 
> Since the Cortex-M7 has D-cache, it would be difficult to work with Cotex-M4 in SMP mode.

Maybe disabling cache will help, but it will decrease its performance.

raiden00pl, 2023-08-14:
@acassis external JLINK works fine with CM4 and is easy to connect (on board 9-Pin Cortex-M header), so I'm not going to waste my time with STlink and openocd. Too bad it doesn't work out-of-box.

In the meantime I managed to get CM4 running and it boots to NSH

acassis, 2023-08-15:
@raiden00pl nice!!! So we can run NuttX on both cores now!

patacongo, 2023-08-15:
> Since the Cortex-M7 has D-cache, it would be difficult to work with Cotex-M4 in SMP mode.

Cache coherency would be an issue as well.  Drivers could not be used on the M4 if they required D-cache operations (flush / invalidate).

> Maybe disabling cache will help, but it will decrease its performance.

By a LOT!  D-cache is the distinguishing feature of the M7 and motivation for its use.

Differing performance would make the behavior non-deterministic.

Looking at the block diagram, the two CPUs are certainly not symmetric.  They connect to the world through different Bus Matrices.  The M7 has access to the M4 AHB Bus, but I don't think that the M4 can access the AXI Bus, except perhaps indirectly via the AHB.  Correct me if I am wrong.  I wonder if the memory maps are the same?

I would be hesitant to consider SMP without a deep understand of the architecture.

raiden00pl, 2023-08-16:
> Looking at the block diagram, the two CPUs are certainly not symmetric. They connect to the world through different Bus Matrices. The M7 has access to the M4 AHB Bus, but I don't think that the M4 [...]

The memory map is the same except each core has its own NVIC.
AXI Bus as you said is available for M4 but not direclty (D1-to-D2 AHB bridge).

> I would be hesitant to consider SMP without a deep understanding of the architecture.
+1

patacongo, 2023-08-18:
> > I would be hesitant to consider SMP without a deep understanding of the architecture.
> 
> +1

I would think that even if the two CPUs had the same memory map, if the loss of deterministic performance,  and if, somehow, the asymmetric D-Cache are not show-stoppers, the performance difference between the two CPUs could make SMP a bad choice.  The M7 is much faster than the M4:  Clocked twice as fast, data-cache, 64-bit AIX, etc.  What?  Probably 2x faster or more?

The performance gains from SMP are usually modest, depending upon the hosted application.  Some applications benefit _a lot_ from SMP, but what you typically see is probably only around a 20% improvement with SMP over a single CPU.  If there is a 50% or more performance loss on one CPU, the result is very likely an overall loss in performance if both CPUs are used SMP, vs. a single CPU.

AMP, on the other hand, always show a performance improvement because you can off-load selected functionality to the M4 to optimize the performance of the M7 application.   In addition, you can get much more interesting low power solutions.

raiden00pl, 2023-08-20:
Quick update: I have openamp working but only with D-cache disabled on CM7. 
At the moment I have no idea what the problem is. Hardware semaphores behave strangely with D-cache enabled and some notifications between cores get lost.  I have `CONFIG_OPENAMP_CACHE=y` but it doesn't help.

patacongo, 2023-08-20:
> Quick update: I have openamp working but only with D-cache disabled on CM7. At the moment I have no idea what the problem is. Hardware semaphores behave strangely with D-cache enabled and some notifications between cores get lost. I have `CONFIG_OPENAMP_CACHE=y` but it doesn't help.

Sounds like some missing cache flush/invalidate logic somewhere.  I stumbled across this:  file:///D:/Spuda/Downloads/an5617-stm32h745755-and-stm32h747757-lines-interprocessor-communications-stmicroelectronics.pdf . It is an application note for using OpenAMP on H7 MCUs.  It does not address cache issues in a lot of depth.  It does recommend setting up the MPU to make the buffer memory non-cacheable.

xiaoxiang781216, 2023-08-20:
> Quick update: I have openamp working but only with D-cache disabled on CM7. At the moment I have no idea what the problem is. Hardware semaphores behave strangely with D-cache enabled and some notifications between cores get lost. I have `CONFIG_OPENAMP_CACHE=y` but it doesn't help.

this option can just handle the top-level buffer (vring and descriptor), so rpmsg buffer need be allocated from uncacheable (but could be bufferable) memory.

raiden00pl, 2023-08-21:
@xiaoxiang781216 @patacongo thanks for the help and save me time looking for the problem. I had read about disabling cache for shared memory before, but completely forgot about it.
After disabling cache for shmem it looks like rptun is working with D-cache enabled:
```
NuttShell (NSH) NuttX-10.4.0
nsh-cm7> rptun ping all 100 100 1 1
current CPU freq: 400000000, ping times: 100
avg: s 0, ns 150837
min: s 0, ns 150150
max: s 0, ns 156002
```

```
NuttShell (NSH) NuttX-10.4.0
nsh-cm4> rptun ping all 100 100 1 1
current CPU freq: 200000000, ping times: 100
avg: s 0, ns 159295
min: s 0, ns 158905
max: s 0, ns 159870
```

closed 2023-08-25 by xiaoxiang781216