apache/nuttx #10239: Error codes related to software flow control in file stm32_hciuart.c

xn365, 2023-08-16:
These should be #ifdef instead of #ifndef：

[https://github.com/apache/nuttx/blob/master/arch/arm/src/stm32/stm32_hciuart.c#L243C1-L245C7](url)
[https://github.com/apache/nuttx/blob/master/arch/arm/src/stm32/stm32_hciuart.c#L269](url)
[https://github.com/apache/nuttx/blob/master/arch/arm/src/stm32/stm32_hciuart.c#L1485C39-L1485C39](url)

This:
https://github.com/apache/nuttx/blob/master/arch/arm/src/stm32/stm32_hciuart.c#L988
`     uin16_t inused = hciuart_rxinuse(config);`

`     if (inuse >= config->rxupper)`
should be :
`     uint16_t inused = hciuart_rxinuse(config);`

`     if (inused >= config->rxupper)`

This:
https://github.com/apache/nuttx/blob/master/arch/arm/src/stm32/stm32_hciuart.c#L1616
`  pinset = (config & ~regval) | GPIO_OUTPUT | GPIO_OUTPUT_SET;`
should be :
`  pinset = (config->rts_gpio & ~regval) | GPIO_OUTPUT | GPIO_OUTPUT_SET;`

acassis, 2023-08-17:
@xn365 good finding! Could you please open a Pull Request?

https://nuttx.apache.org/docs/latest/contributing/index.html

casaroli, 2026-11-08:
Fixed by #10360.

closed 2026-11-08 by casaroli