apache/nuttx #10253: Make configure error more user friendly

acassis, 2023-08-17:
Currently if the users forget to clone the apps/ at same level of nuttx/ he/she will receive this error:
```
$ ./tools/configure.sh sim:nsh
  Copy files
  Select CONFIG_HOST_LINUX=y
  Refreshing...
CP: arch/dummy/Kconfig to /tmp/nuttx/arch/dummy/dummy_kconfig
CP: boards/dummy/Kconfig to /tmp/nuttx/boards/dummy/dummy_kconfig
LN: include/arch to arch/sim/include
LN: include/arch/board to /tmp/nuttx/boards/sim/sim/sim/include
LN: drivers/platform to /tmp/nuttx/drivers/dummy
LN: include/arch/chip to /tmp/nuttx/arch/sim/include/sim
LN: arch/sim/src/chip to /tmp/nuttx/arch/sim/src/sim
LN: arch/sim/src/board to /tmp/nuttx/boards/sim/sim/sim/src
Kconfig:2299: can't open file "/Kconfig"
make: *** [tools/Unix.mk:680: olddefconfig] Error 1
ERROR: failed to refresh
```

It should be important to detect this error and explain to user that the ../apps directory is missing and that it needs to be cloned there

danielappiagyei-bc, 2023-09-10:
Hi @acassis ,
Thank you for making this issue! I actually ran into this and was stuck for a bit until I searched for "failed to refresh" on here and found this post.

I was following step 4 of this [guide](https://nuttx.apache.org/docs/latest/contributing/making-changes.html) and made the mistake of not naming my cloned repos "nuttx" and "apps". Instead, I gave them the name of my forked nuttx repo. (`nuttx-forked`, `nuttx-apps-forked`)
In other words, I did:

```
git clone <your forked nuttx project clone url>
git clone <your forked nuttx-apps project clone url>
```

instead of 

```
git clone <your forked nuttx project clone url> nuttx
git clone <your forked nuttx-apps project clone url> apps
```

I think a note under step 4 would be helpful to future users:
"""
You **must** name the cloned repos "**nuttx**" and "**apps**", respectively, and they **must** reside in the same directory. Failure to do so will result in an error when running `.tools/configure.sh` similar to below:
```
Kconfig:2299: can't open file "/Kconfig"
make: *** [tools/Unix.mk:680: olddefconfig] Error 1
ERROR: failed to refresh
```
"""

Thanks!

xuxin930, 2023-09-22:
hi @acassis great suggestion.
I think this's part of the reason:
https://github.com/apache/nuttx/blob/7f574210a73e5fb44f3323b7ba875958de4de5d0/tools/configure.sh#L282-L288
we only considered the case where posappdir does not exist.
However, when the apps directory is not found, the directory checked is `nuttx/tools/..`, which will always exist.
I think we should check whether it exists when appdir is assigned a value earlier, at LINE273
https://github.com/apache/nuttx/blob/7f574210a73e5fb44f3323b7ba875958de4de5d0/tools/configure.sh#L255-L275

closed 2023-09-22 by xiaoxiang781216