apache/nuttx #10388: libboard.a is not generated by make export

trns1997, 2023-08-24:
I encountered this issue while calling "make export" immediately after configuring my board application. It appears that "libboard.a" is not being built or copied to the destination folder before generating the export.

## How to reproduce
```
P:➜  nuttx git:(master) ✗ tools/configure.sh -E -l xmc4700-relax:nsh
make: Entering directory '/home/trns1997/nuttxspace/nuttx'
make: Leaving directory '/home/trns1997/nuttxspace/nuttx'
  Copy files
  Select CONFIG_HOST_LINUX=y
  Refreshing...
CP: arch/dummy/Kconfig to /home/trns1997/nuttxspace/nuttx/arch/dummy/dummy_kconfig
CP: boards/dummy/Kconfig to /home/trns1997/nuttxspace/nuttx/boards/dummy/dummy_kconfig
LN: platform/board to /home/trns1997/nuttxspace/apps/platform/dummy
LN: include/arch to arch/arm/include
LN: include/arch/board to /home/trns1997/nuttxspace/nuttx/boards/arm/xmc4/xmc4700-relax/include
LN: drivers/platform to /home/trns1997/nuttxspace/nuttx/drivers/dummy
LN: include/arch/chip to /home/trns1997/nuttxspace/nuttx/arch/arm/include/xmc4
LN: arch/arm/src/chip to /home/trns1997/nuttxspace/nuttx/arch/arm/src/xmc4
LN: arch/arm/src/board to /home/trns1997/nuttxspace/nuttx/boards/arm/xmc4/xmc4700-relax/src
mkkconfig in /home/trns1997/nuttxspace/apps/audioutils
mkkconfig in /home/trns1997/nuttxspace/apps/benchmarks
mkkconfig in /home/trns1997/nuttxspace/apps/boot
mkkconfig in /home/trns1997/nuttxspace/apps/canutils
mkkconfig in /home/trns1997/nuttxspace/apps/crypto
mkkconfig in /home/trns1997/nuttxspace/apps/examples/mcuboot
mkkconfig in /home/trns1997/nuttxspace/apps/examples
mkkconfig in /home/trns1997/nuttxspace/apps/fsutils
mkkconfig in /home/trns1997/nuttxspace/apps/games
mkkconfig in /home/trns1997/nuttxspace/apps/graphics
mkkconfig in /home/trns1997/nuttxspace/apps/industry
mkkconfig in /home/trns1997/nuttxspace/apps/interpreters/luamodules
mkkconfig in /home/trns1997/nuttxspace/apps/interpreters
mkkconfig in /home/trns1997/nuttxspace/apps/logging
mkkconfig in /home/trns1997/nuttxspace/apps/lte
mkkconfig in /home/trns1997/nuttxspace/apps/math
mkkconfig in /home/trns1997/nuttxspace/apps/mlearning
mkkconfig in /home/trns1997/nuttxspace/apps/netutils
mkkconfig in /home/trns1997/nuttxspace/apps/sdr
mkkconfig in /home/trns1997/nuttxspace/apps/system
mkkconfig in /home/trns1997/nuttxspace/apps/testing
mkkconfig in /home/trns1997/nuttxspace/apps/wireless/bluetooth
mkkconfig in /home/trns1997/nuttxspace/apps/wireless/ieee802154
mkkconfig in /home/trns1997/nuttxspace/apps/wireless
mkkconfig in /home/trns1997/nuttxspace/apps
Loaded configuration '.config'
Configuration saved to '.config'
```
```
P:➜  nuttx git:(master) ✗ make export
Create version.h
LN: platform/board to /home/trns1997/apps/platform/dummy
Register: nsh
Register: sh
Register: ramtest
```
### Result
```
P:➜  nuttx git:(master) ✗ tar -xzf nuttx-export-12.2.1.tar.gz && ls nuttx-export-12.2.1/libs   
libapps.a  libarch.a  libbinfmt.a  libboards.a  libc.a  libdrivers.a  libfs.a  libmm.a  libsched.a
```
`libboard.a` is missing in the exported archive.

## Current Work around
Call `make` before calling `make export`
```
P:➜  nuttx git:(master) ✗ tools/configure.sh -E -l xmc4700-relax:nsh
make: Entering directory '/home/trns1997/nuttxspace/nuttx'
make: Leaving directory '/home/trns1997/nuttxspace/nuttx'
  Copy files
  Select CONFIG_HOST_LINUX=y
  Refreshing...
CP: arch/dummy/Kconfig to /home/trns1997/nuttxspace/nuttx/arch/dummy/dummy_kconfig
CP: boards/dummy/Kconfig to /home/trns1997/nuttxspace/nuttx/boards/dummy/dummy_kconfig
LN: platform/board to /home/trns1997/nuttxspace/apps/platform/dummy
LN: include/arch to arch/arm/include
LN: include/arch/board to /home/trns1997/nuttxspace/nuttx/boards/arm/xmc4/xmc4700-relax/include
LN: drivers/platform to /home/trns1997/nuttxspace/nuttx/drivers/dummy
LN: include/arch/chip to /home/trns1997/nuttxspace/nuttx/arch/arm/include/xmc4
LN: arch/arm/src/chip to /home/trns1997/nuttxspace/nuttx/arch/arm/src/xmc4
LN: arch/arm/src/board to /home/trns1997/nuttxspace/nuttx/boards/arm/xmc4/xmc4700-relax/src
mkkconfig in /home/trns1997/nuttxspace/apps/audioutils
mkkconfig in /home/trns1997/nuttxspace/apps/benchmarks
mkkconfig in /home/trns1997/nuttxspace/apps/boot
mkkconfig in /home/trns1997/nuttxspace/apps/canutils
mkkconfig in /home/trns1997/nuttxspace/apps/crypto
mkkconfig in /home/trns1997/nuttxspace/apps/examples/mcuboot
mkkconfig in /home/trns1997/nuttxspace/apps/examples
mkkconfig in /home/trns1997/nuttxspace/apps/fsutils
mkkconfig in /home/trns1997/nuttxspace/apps/games
mkkconfig in /home/trns1997/nuttxspace/apps/graphics
mkkconfig in /home/trns1997/nuttxspace/apps/industry
mkkconfig in /home/trns1997/nuttxspace/apps/interpreters/luamodules
mkkconfig in /home/trns1997/nuttxspace/apps/interpreters
mkkconfig in /home/trns1997/nuttxspace/apps/logging
mkkconfig in /home/trns1997/nuttxspace/apps/lte
mkkconfig in /home/trns1997/nuttxspace/apps/math
mkkconfig in /home/trns1997/nuttxspace/apps/mlearning
mkkconfig in /home/trns1997/nuttxspace/apps/netutils
mkkconfig in /home/trns1997/nuttxspace/apps/sdr
mkkconfig in /home/trns1997/nuttxspace/apps/system
mkkconfig in /home/trns1997/nuttxspace/apps/testing
mkkconfig in /home/trns1997/nuttxspace/apps/wireless/bluetooth
mkkconfig in /home/trns1997/nuttxspace/apps/wireless/ieee802154
mkkconfig in /home/trns1997/nuttxspace/apps/wireless
mkkconfig in /home/trns1997/nuttxspace/apps
Loaded configuration '.config'
Configuration saved to '.config'
```
```
P:➜  nuttx git:(master) ✗ make && make export
Create version.h
LN: platform/board to /home/trns1997/apps/platform/dummy
Register: nsh
Register: sh
Register: ramtest
LD: nuttx
CP: nuttx.hex
CP: nuttx.bin
```
### Result
```
P:➜  nuttx git:(master) ✗ tar -xzf nuttx-export-12.2.1.tar.gz && ls nuttx-export-12.2.1/libs
libapps.a  libarch.a  libbinfmt.a  libboard.a  libboards.a  libc.a  libdrivers.a  libfs.a  libmm.a  libsched.a
```

As you can see the `libboard.a` was correctly generated and packaged in the second case. I will investigate on my end to see what is causing the issue.

acassis, 2023-08-24:
@xiaoxiang781216 do you have "make export" in the CI? I think it is important for some customers that aren't using the NuttX building system.

xiaoxiang781216, 2023-08-24:
No, ci just invoke make now, it could be changed to "make export" and "make import".

trns1997, 2023-08-28:
@acassis @xiaoxiang781216,

After some investigation, I've identified the reason behind the absence of the `libboard.a` packaging. The issue lies within the file `tools/mkexport.sh`, specifically at line 354. This script checks for the existence of the `libboard.a` file and packages it accordingly.

I looked into the makefiles and found that `FlatLibs.mk` requires the `CONFIG_ARCH_BOARD_COMMON` option to be set. This addition enables the inclusion of `libboard` in the `EXPORTLIBS` list. Consequently, I resolved this problem by setting the `CONFIG_ARCH_BOARD_COMMON` flag within my board's `defconfig`, effectively resolving the issue.

However, I have a query in relation to this matter:

1. Wouldn't it be prudent for all boards in the NuttX project to default to `CONFIG_ARCH_BOARD_COMMON=y` in their respective `defconfig` files? Such an approach would enable immediate execution of apps on Board X. Should a user possess a custom board utilizing the same MCU, they could then deactivate this flag and furnish the necessary functions for their project's functionality.

2. Regarding the behavior where calling `make` prior to `make export` results in the generation of `libboard.a`, leading to the successful discovery and packaging of the file by the `tools/mkexport.sh` script: Is this intended behavior? It seems conceivable that a user might inadvertently execute `make export` after generating `libboard.a`, potentially leading to its inclusion in the export. This situation might not align with the intended behavior, especially if the `CONFIG_ARCH_BOARD_COMMON` flag is deactivated.

My proposition would be maybe to consider invoking `make clean` before `make export`. This approach would ensure the export process incorporates the correct libraries, aligning with user expectations. Let me know what you guys think.

trns1997, 2025-09-17:
Hi @acassis @xiaoxiang781216,
I would like to readdress this forgotten issue (srry i for got about it). This https://github.com/apache/nuttx/issues/17024 made me rediscover this issue. I also saw that this PR https://github.com/apache/nuttx/pull/16241and issue https://github.com/apache/nuttx/issues/12122 tried to mitigate a similar situation by setting `CONFIG_ARCH_BOARD_COMMON=y` for some boards, but I’m unsure what the expected behavior is here:

- Should make export automatically build libboard.a as part of the export process, even if no prior make was run?
- Or is this board-specific and requires CONFIG_ARCH_BOARD_COMMON to be set in the defconfig by the user?
- If this is expected, should make export fail early with a clearer message if libboard.a is missing rather than silently creating an incomplete package?

Would appreciate some guidance on what’s considered correct so I can help investigate or fix this properly.

trns1997, 2025-09-17:
Another question is it a good idea to add to all boards defconfig with board common to activate  CONFIG_ARCH_BOARD_COMMON=y by default?