apache/nuttx #10416: How to print filename and line by assert

SPRESENSE, 2023-08-28:
`CONFIG_DEBUG_ASSERTIONS_FILENAME` must be enabled in order to print filename and line by assert function.
And, `CONFIG_DEBUG_ASSERTIONS_FILENAME` depends on `CONFIG_DEBUG_ASSERTIONS` and `CONFIG_DEBUG`.
So `assert()` and `ASSERT()` macro, not `DEBUGASSERT()`, cannot output the filename and line if `CONFIG_DEBUG=n`.

Is this behavior as intended?
It is strange because there is no distinction between `ASSERT()` and `DEBUGASSERT()`.

SPRESENSE, 2023-08-28:
Introduced by https://github.com/apache/nuttx/pull/9498

xiaoxiang781216, 2023-08-29:
so you want to remove the dependence?

SPRESENSE, 2023-08-30:
We should not only remove the dependencies, but rename the configration name.
Or, it is a good idea to create a separate `CONFIG_ASSERTIONS_FILENAME` from `CONFIG_DEBUG_ASSERTIONS_FILENAME` and apply it to `assert()`, `ASSERT()` and `PANIC()`.

anchao, 2023-10-11:
I just check the config, `CONFIG_DEBUG_ASSERTIONS_FILENAME` is default y if enable `CONFIG_DEBUG_ASSERTIONS` ,
the assert file name could be printed if enable `CONFIG_DEBUG_ASSERTIONS` and call `assert()`, `ASSERT() `and `PANIC().`

I don't quite understand what your problem is. Do you want to enable `CONFIG_DEBUG_ASSERTIONS_FILENAME` without  `CONFIG_DEBUG_ASSERTIONS`?

SPRESENSE, 2023-10-11:
I think that `assert()` should print the filename and line without `CONFIG_DEBUG`.

SPRESENSE, 2023-10-18:
@anchao
> You mean CONFIG_DEBUG is actually CONFIG_NDEBUG=y right?

No. It is not about `CONFIG_NDEBUG`.

The NuttX assertion has two types, `assert()`(=`ASSERT()`) and `DEBUGASSERT()`.
The former is used standardly by applications.
The latter is used for special debugging purposes.
To prevent an increase of the code size, the `DEBUGASSERT()` code can be disabled by `CONFIG_DEBUG_ASSERTIONS=n`.

| Function         |CONFIG                   | CONFIG                           | Show filename/line |
| ---              | ---                     | ---                              | ---                    |
|assert(), ASSERT()|CONFIG_DEBUG_ASSERTIONS=n|                                  | No(*1)                 |
|DEBUGASSERT()     |CONFIG_DEBUG_ASSERTIONS=y|CONFIG_DEBUG_ASSERTIONS_FILENAME=n| No                     |
|DEBUGASSERT()     |CONFIG_DEBUG_ASSERTIONS=y|CONFIG_DEBUG_ASSERTIONS_FILENAME=y| Yes                    |

(*1)
When the general application uses `assert()`, the filename and line are not shown.
Before your commit, the filename and line were shown.

This comes from the following code.
https://github.com/apache/nuttx/blob/master/include/assert.h#L45-L51

For `assert()` to print the filename and line, we need to enable `CONFIG_DEBUG=y`, `CONFIG_DEBUG_ASSERTIONS=y` and `CONFIG_DEBUG_ASSERTIONS_FILENAME=y`. It causes an increase in code size since `DEBUGASSERT()` is also enabled.

Essentially, since `assert()` is also used by the application, it should show the filename and line by default, as it does on other platforms.

How do you think?

anchao, 2023-10-19:
Thanks for the clear explanation, I followed your suggestion to separate CONFIG_ASSERTIONS_FILENAME from CONFIG_DEBUG_ASSERTIONS_FILENAME, and remove the coupling between assert(3) with CONFIG_DEBUG_ASSERTIONS,
Please review PR https://github.com/apache/nuttx/pull/10978