apache/nuttx #10433: Suggestion: Unified usbdev enumeration process

zyfeier, 2023-08-29:
In the current usbdev code, each class driver and composite driver
has its own enumeration process, resulting in a high degree of code
duplication and making maintenance difficult.

Maybe we can abstract out the common processes, similar to the composite
device framework in Linux. Different functions only need to provide
descriptor information, and the common processes will be handled by
the composite framework.

xiaoxiang781216, 2023-08-29:
@raiden00pl how about your opinion?

raiden00pl, 2023-08-29:
There is a lot of duplicate code and logic in usbdevs and it is not well unified. This makes debugging and reviewing difficult.
Separating the common code is a good idea, but I'm not sure about calling this common USB device logic `composite`.

xiaoxiang781216, 2023-08-29:
it depends on how you interpret composite. The single function device can be considering a special case of composition which just has one function. Since many common code is duplicated in each function driver, the benefit of consolidating the common to composite driver could:

1.  Remove the source code duplication
2. Reduce the final image code size

Linux: https://github.com/torvalds/linux/tree/master/drivers/usb/gadget

xiaoxiang781216, 2023-08-31:
Ok, so your concern is the naming, not the unification. How about we do the unification first and then you can choice a better name to do search and replace in the batch?

raiden00pl, 2023-08-31:
> Ok, so your concern is the naming, not the unification. How about we do the unification first and then you can choice a better name to do search and replace in the batch?

I'm ok with that approach.
If there are no more objections to the naming than mine, then I won't insist on changing naming either.

closed 2023-09-20 by xiaoxiang781216