apache/nuttx #10445: how to format spi flash as vfat

xiaotailang, 2023-08-30:
How can I format an SPI flash as a VFAT file system and manage it through the MTD interface? Are there any reference examples? Currently, I only see the method of formatting it as NXFFS and managing it through MTD.

xiaoxiang781216, 2023-08-30:
no, vfat can't work with mtd directly (mtd is totally different storage from the tradition block device). You can either:

1. Use the flash friendly file system(e.g. littlefs, nxffs, smartfs)
2. Utilize ftl(e.g.  which convert mtd interface to block interface

xiaotailang, 2023-08-30:
Thank you very much. Are there any reference examples for using other file systems on SPI flash, such as LittleFS or SmartFS? I have encountered some issues while using NXFFS.

xiaoxiang781216, 2023-08-30:
you can search the related config in nuttx/boards.

xiaotailang, 2023-09-11:
@xiaoxiang781216 Hello, when I was testing the file system, I tried using littlefs and it worked fine. However, when I attempted to use smartfs and referred to some boards, I encountered some issues. Here are the steps I followed:
  CONFIG_FS_SMARTFS=y
  CONFIG_SMARTFS_ERASEDSTATE=0xff
  CONFIG_SMARTFS_MAXNAMLEN=16

  CONFIG_MTD_SMART=y
  CONFIG_MTD_SMART_SECTOR_SIZE=512
  CONFIG_MTD_SMART_WEAR_LEVEL=y

Next, I added the following code:
smart_initialize(minor, mtd, "spiw25");

if (ret < 0)
  {
    finfo("smart_initialize failed, Trying to erase first...\n");
    ret = mtd->ioctl(mtd, MTDIOC_BULKERASE, 0);
    if (ret < 0)
      {
        ferr("ERROR: ioctl(BULKERASE) failed: %d\n", ret);
        return ret;
      }

    finfo("Erase successful, initializing it again.\n");
    smart_initialize(minor, mtd, "spiw25");
    if (ret < 0)
      {
        ferr("ERROR: smart_initialize failed: %d\n", ret);
        return ret;
      }
  }

When I entered "mksmartfs /dev/smart0," there was no error, but there was no "smart0" file generated in the "/dev/" directory. Consequently, the subsequent "mount" operation failed. Are there any additional configurations or steps that I may have overlooked?

TimJTi, 2023-09-11:
I think you need something like (where mnt_name is your choice)?

  ret = nx_mount("/dev/smart0", "/mnt/mnt_name", "smartfs", 0, autoformat);

xiaotailang, 2023-09-11:
@TimJTi 

> I think you need something like (where mnt_name is your choice)?
> 
> ret = nx_mount("/dev/smart0", "/mnt/mnt_name", "smartfs", 0, autoformat);

Yes, I have resolved the issue.。
I just tried the flash_eraseall command and am able to fully erase the SPI flash. Have you ever encountered a problem where writing to a SPI flash formatted with littlefs would get stuck when trying to write a file using the echo "12345" > 1.txt command during your previous development process? I am seeing this issue now while testing on top of littlefs, and I suspect it might be due to the echo command not interfacing correctly with littlefs.

TimJTi, 2023-09-11:
Ok, away at moment so GitHub phone app sometimes slow to refresh. Sorry if I answered a question that didn't need answers!

I only ever tried things like "ls -l > /mnt/flash" and that always worked as long as the memory was properly formatted and mounted. I tried vfat, littlefs, smartfs and nxffs. Settled on littlefs in the end.

xiaotailang, 2023-09-11:
> Ok, away at moment so GitHub phone app sometimes slow to refresh. Sorry if I answered a question that didn't need answers!
>
> I only ever tried things like "ls -l > /mnt/flash" and that always worked [...]
ok ! thank you very much for your response.

TimJTi, 2023-09-11:
Should have been /mnt/flash/file.txt or similar of course. I think explicit paths may always be needed too? Or, at least, ./file.txt

xiaotailang, 2023-09-12:
> Should have been /mnt/flash/file.txt or similar of course. I think explicit paths may always be needed too? Or, at least, ./file.txt

ok !Thank you once again.

closed 2023-09-20 by xiaoxiang781216