apache/nuttx #10454: to support LIN ,   how about to add a Individual character driver

xucheng5, 2023-08-31:
refer to https://www.mail-archive.com/dev@nuttx.apache.org/msg06818.html & https://github.com/apache/nuttx/pull/5043

a LIN upper-half character driver :   will do : 
1.  use a lin_frame as a uint of transmit
    Send :  
            LIN master task :    header send & repsonse send is necessary ;
            LIN slave  task   :    only repoonse send .
    Recv :
            both master & slave always recv frame from lin bus.

    for Send & Recv reponse,    because uart use char as u uint of transmit , so use struct lin_frame_s as a uint of transmit could be 
    good choice
    
    `struct lin_frame_s 
     {
         u8 pid;
         u8 data[MAX_LIN_LEN] ;  // MAX_LIN_LEN = 8;
     }
   

     
2.  need a  LIN filter  which  apps can catch specific PID - LIN frame,  like  can  characterdriver.
3.  A SW implementation for  State Machine ，break check,   frame error,  crc check  ( like  [sllin.c - Pavel Pisa](https://github.com/ppisa/linux-lin/tree/master/sllin)).

pkarashchenko, 2023-08-31:
I had a SocketLIN in my personal backlog for at least half a year ;) Let me see if I can prioritize this activity and get some expected dates.

xucheng5, 2023-08-31:
> I had a SocketLIN in my personal backlog for at least half a year ;) Let me see if I can prioritize this activity and get some expected dates.

SocketLIN will also be good choice !

xiaoxiang781216, 2023-08-31:
@pkarashchenko it's good that you can upstream SocketLIN, we help review the change and develop a real driver to test it.

TimJTi, 2023-08-31:
I have LIN on my custom board, and a CAN/LIN sniffer/analyser so I will certainly be able to help with testing.

xiaoxiang781216, 2023-09-07:
@pkarashchenko could you spend some time to provide the SocketLIN driver? Thanks. We want to develop a LIN driver base on your work.

pkarashchenko, 2023-09-07:
Hi. I need to build here a schedule a bit as I will be most probably able to handle this activity only at the second part of October.

xiaoxiang781216, 2023-09-19:
@pkarashchenko could you send out a draft PR? so we can polish your work.

xiaoxiang781216, 2023-10-10:
@pkarashchenko do you have any update about SocketLIN?

pkarashchenko, 2023-10-23:
@xiaoxiang781216 I'm not sure if that is what you expected, but here is what I've been able to conduct till now: https://github.com/apache/nuttx/pull/11002
Similar to Linux this code relies on the SocketCAN upper layer infrastructure. I've backported that from the kind of "selfmade" LIN driver and going to verify that SocketLIN driver will still be working as expected after backport. Till now I just verified that LIN interfaces are registered and I can configure it via `ifup linX`. I hope to pass send/recv messages test till the end of the week.

pkarashchenko, 2023-10-24:
There are few differences to discuss about SocketLIN:
1. Linux slin use some extra bits in CAN-ID to "enable cache response" or "request extended checksum"
```
#define LIN_CANFR_FLAGS_OFFS	6 /* Lower 6 bits in can_id correspond to LIN ID */

#define LIN_CACHE_RESPONSE	(1 << (LIN_CANFR_FLAGS_OFFS))
#define LIN_CHECKSUM_EXTENDED	(1 << (LIN_CANFR_FLAGS_OFFS + 1))
#define LIN_SINGLE_RESPONSE (1 << (LIN_CANFR_FLAGS_OFFS + 2))
```
as well as for error reporting
```
/* Error flags */
#define LIN_ERR_RX_TIMEOUT	(1 << (LIN_CANFR_FLAGS_OFFS + 8))
#define LIN_ERR_CHECKSUM	(1 << (LIN_CANFR_FLAGS_OFFS + 9))
#define LIN_ERR_FRAMING		(1 << (LIN_CANFR_FLAGS_OFFS + 10))
```
while currently in my draft I used `CAN_EFF_FLAG` for selection of extended vs classical checksum,
2. slin uses `CAN_EFF_FLAG` flag for configure frame cache
```
#define LIN_CTRL_FRAME 		CAN_EFF_FLAG
```
and in my PR I use TX frames as a regular frames while RX frames are always done via RTR.

I suppose that frame caching is needed for LIN slave mode as https://github.com/apache/nuttx/pull/11002 adds only LIN master support. Adding a LIN slave may require slight rework as all TX data should be cached until master requests with a proper LIN-ID, so in CAN language "we will need COB (communication object)" to be allocated for each TX ID in case of LIN slave.

closed 2024-09-19 by xiaoxiang781216