apache/nuttx #10479: LED_PANIC never used?

Siviuze, 2023-09-04:
Hello,

I tried to use the LED_PANIC feature but it is not working... Is there a logic that must be implemented to handle this case? I cannot find any call to board_autoled_on() with it in the code base.

Note: I provoke a panic with the TTY_FORCE_PANIC feature.

acassis, 2023-09-05:
@Siviuze It was working (everything we are implementing a driver and it crash the kernel the LED starts blinking, at leat onSTM32). You just need to have ARCH_LEDS, please see as it is implemented on stm32f103-minimum and other boards.

trns1997, 2023-09-06:
@acassis I can confirm that it does not work, I tested on an `stm32f411e-disco` board it required a couple of modification but the panic led does not blink.

acassis, 2023-09-07:
@trns1997 you are right:

```
nuttx/boards/arm/stm32/stm32f4discovery$ git grep ARCH_LED
configs/pm/defconfig:# CONFIG_ARCH_LEDS is not set
include/board.h:/* If CONFIG_ARCH_LEDS is not defined, then the user can control the LEDs
include/board.h:/* If CONFIG_ARCH_LEDs is defined, then NuttX will control the 4 LEDs on
src/CMakeLists.txt:if(CONFIG_ARCH_LEDS)
src/Make.defs:ifeq ($(CONFIG_ARCH_LEDS),y)
src/stm32_autoleds.c:#ifdef CONFIG_ARCH_LEDS
src/stm32_autoleds.c:#endif /* CONFIG_ARCH_LEDS */
src/stm32_boot.c:#ifdef CONFIG_ARCH_LEDS
src/stm32_idle.c:#if defined(CONFIG_ARCH_LEDS) && defined(LED_IDLE)
src/stm32_userleds.c:#ifndef CONFIG_ARCH_LEDS
src/stm32_userleds.c:#endif /* !CONFIG_ARCH_LEDS */

nuttx/boards/arm/stm32/stm32f411e-disco$ git grep ARCH_LED
configs/nsh/defconfig:# CONFIG_ARCH_LEDS is not set
src/stm32_boot.c:#ifdef CONFIG_ARCH_LEDS
alan@dev:~/nuttxspace/nuttx/boards/arm/stm32/stm32f411e-disco$ 
```

Just copy the code from stm32f4discovery to your board

trns1997, 2023-09-08:
@acassis the problem is that no where in the code base we see a place that calls `board_autoled_on(LED_PANIC)`, there fore the `LED_PANIC` does not blink ever. There seems to be something missing. Or we misunderstood something.

TimJTi, 2023-09-09:
PR7875 is to blame I think. No proper regression testing perhaps? What else got broken with such a huge number of files changed?

trns1997, 2023-09-09:
@TimJTi i think along with that the problem started in this particular PR#7808. Now what else is broken i am not sure. I guess we'll find out as people test and try to use these features.

PR #10553 (2023-09-20): fix led panic feature
## Summary
- Fix LED PANIC Feature to blink at 2Hz.
- Closes #10479

## Impact

## Testing