apache/nuttx #10488: Assessment of the difficulty in porting CPU architecture for nuttx

wangyuliu, 2023-09-06:
Hello everyone! I am working on implementing a tool to assess the complexity of CPU architecture porting. It primarily focuses on RISC-V architecture porting. As part of my dataset, I have collected the nuttx project. By using scanning tools, I have determined that the porting complexity is high due to a significant amount of code related to the CPU architecture. It would require a professional team to handle this task. Is this assessment accurate? I look forward to your help and response.
各位好！我在实现一个关于CPU架构的移植工作复杂度评估的工具，主要面向riscv架构移植，我收集nuttx项目作为我的数据集之一，通过扫描工具得出其移植复杂度较大，工程中与CPU架构相关的代码量多，需要较专业团队进行，这样的判断是否准确？期待您的帮助与回答

wangyuliu, 2023-09-06:
referring to the overall workload from adapting the project to a specific architecture to achieving full functionality on that architecture

xiaoxiang781216, 2023-09-06:
but riscv is already supported.

wangyuliu, 2023-09-06:
Sorry, I may not have described the issue clearly. What I want to ask is the overall workload of porting to a certain architecture, such as RisCV architecture. This work may have been completed, but it does not conflict. Thank you for your answer

jlaitine, 2023-09-06:
Porting *any* full operating system to a new CPU architecture is a complex task and requires highly skilled professional team - although for NuttX it is possible to do by even by just one skilled individual (there are examples of that).

It is actually more fruitful to compare the complexity of porting of different operating systems, *with equal feature set* to a new processor architecture.

Personally I find NuttX being quite easy to port to new devices and even to new CPU architectures. Although as Xiao said, RISC-V is already supported, like most of the common CPU architectures.

Porting to a new architecture is quite straightforward because NuttX already supports so many different architectures. The code is already well organized and CPU architecture specific parts are well separated from the common code.

It is not possible to say anything about the accuracy of your assessment without knowing the full process and data used as a basis of the assessment. For example, as I mentioned earlier, complex compared to what? Which other OS's have you assessed and what are the datasets and key indicators?

acassis, 2023-09-07:
@wangyuliu what RISC-V CPU/MCU are you planing to port NuttX for?

closed 2023-09-20 by xiaoxiang781216

wangyuliu, 2023-12-05:
> Porting _any_ full operating system to a new CPU architecture is a complex task and requires highly skilled professional team - although for NuttX it is possible to do by even by just one skilled [...]

Our tool evaluated these projects and judged them to be of medium complexity: rt-thread, freeRTOS, uC-OS2, zephyr, etc. Projects with high portability complexity include gcc, openCV, binutils-gdb, etc., while projects with low portability complexity are similar to sqlite. The reason why our tool analyzes nuttx and calls it medium transplant complexity is that a large amount of assembly code is scanned out in the project. We judge it as a part that needs to be modified during the transplant process.  Let’s analyze it based on this factor,Is it okay to have a medium porting complexity for nuttx?I'm looking forward to your answer.