apache/nuttx #10556: Segger Real Time Transfer (RTT) Usage

danielappiagyei-bc, 2023-09-09:
Hi,
Is there any documentation or examples on how to use Segger's RTT functionality in NuttX?
I've read [segger's documentation](https://wiki.segger.com/RTT) and set the appropriate [nuttx kconfig's](https://github.com/apache/nuttx/blob/nuttx-12.2.1/drivers/segger/Kconfig):

```
CONFIG_STREAM_RTT=y
CONFIG_SEGGER_RTT=y
# IMXRT1064 cache line is 4 bytes
CONFIG_SEGGER_RTT_CPU_CACHE_LINE_SIZE=4
```

After compiling I see an `RTT` directory generated under `drivers/segger/` with segger's source and header RTT implementation files.
I include [segger/rtt.h](https://github.com/apache/nuttx/blob/nuttx-12.2.1/include/nuttx/segger/rtt.h) in a file of mine that gets compiled **as C++** but compilation fails due to `rtt.h` using the C++ reserved keyword `public` for fields in `struct lib_rttoutstream_s` and `struct lib_rttinstream_s`. Additionally, the transitive include [streams.h](https://github.com/apache/nuttx/blob/nuttx-12.2.1/include/nuttx/streams.h) uses `public` as well as the `this` keyword (e.g., [here](https://github.com/apache/nuttx/blob/nuttx-12.2.1/include/nuttx/streams.h#L73) and [here](https://github.com/apache/nuttx/blob/nuttx-12.2.1/include/nuttx/streams.h#L140)).

rtt.h uses a '#ifdef __cplusplus' guard [here](https://github.com/apache/nuttx/blob/nuttx-12.2.1/include/nuttx/segger/rtt.h#L62) so I'm assuming including it from a c++ file was an intended use case. Could someone help me understand if this is a bug or if I'm attempting to use the RTT API incorrectly?

Thanks!

**NuttX version**: 12.2.1

xiaoxiang781216, 2023-09-10:
> Hi, Is there any documentation or examples on how to use Segger's RTT functionality in NuttX? I've read [segger's documentation](https://wiki.segger.com/RTT) and set the appropriate [nuttx [...]

In most case, you don't need use rtt stream in the code directly, rtt already integrate into various nuttx subsystem:

1. semihost(access PC file): enable CONFIG_FS_HOSTFS and ARM_SEMIHOSTING_HOSTFS, issue this command in nsh:
```
    mount -t hostfs -o fs=/path/to/host /path/to/nuttx
```
2. sysview: https://github.com/apache/nuttx/issues/5843 https://github.com/apache/nuttx/issues/9991
3. thread awareness: https://github.com/apache/nuttx/issues/9200
4. rtt serial console: https://github.com/apache/nuttx/pull/9777
5. rtt syslog: https://github.com/apache/nuttx/pull/5388

> I include [segger/rtt.h](https://github.com/apache/nuttx/blob/nuttx-12.2.1/include/nuttx/segger/rtt.h) in a file of mine that gets compiled **as C++** but compilation fails due to `rtt.h` using the [...]

it's wrong to use public(C++ keyword) in stream interface, please provide a patch to rename public to other word(e.g. common).

danielappiagyei-bc, 2023-09-10:
> it's wrong to use public(C++ keyword) in stream interface, please provide a patch to rename public to other word(e.g. common).

Should I also provide a patch for the `this` keyword being used? Also would forbidding the use of C++ keywords in C be something to add to the [C coding standard](https://nuttx.apache.org/docs/latest/contributing/coding_style.html) as well as automated checks in `nxstyle`? Not sure if this is something that should be brought up in the mailing list or not.

> 4. rtt serial console: https://github.com/apache/nuttx/pull/9777

Thanks Xiang, it looks like this is equivalent to what I was looking for. I see that this PR introduced the `SERIAL_RTT` kconfig.
This was after NuttX 12.2.1 was released so it didn't make the cut.
For posterity, I'll post a code snippet of how I was able to get RTT working in my nuttx version in a C file:

```
  #include <nuttx/segger/rtt.h>
  ....

  struct lib_rttoutstream_s out_stream = {0};
 /* channel 0 is reserved by RTT for the Terminal*/
  lib_rttoutstream_open(&out_stream, 0, CONFIG_SEGGER_RTT_BUFFER_SIZE_UP);

  /* the below should be equivalent to `out_stream.public.puts(&out_stream.public, "dan\r\n", sizeof("dan\r\n");`  */
  out_stream.public.putc(&out_stream.public, (int)'d');
  out_stream.public.putc(&out_stream.public, (int)'a');
  out_stream.public.putc(&out_stream.public, (int)'n');
  out_stream.public.putc(&out_stream.public, (int)'\r');
  out_stream.public.putc(&out_stream.public, (int)'\n');
  out_stream.public.putc(&out_stream.public, (int)'\0');

  while (1){
        usleep(500);
    }

```

Using Segger's [JLinkRTTViewer](https://www.segger.com/products/debug-probes/j-link/tools/rtt-viewer/), I attached to the existing J-link debug session I had with Segger Ozone and was able to see my log in Terminal 0: 
[image]

danielappiagyei-bc, 2023-09-10:
> I include [segger/rtt.h](https://github.com/apache/nuttx/blob/nuttx-12.2.1/include/nuttx/segger/rtt.h) in a file of mine that gets compiled as C++ but compilation fails due to rtt.h using the C++ [...]

Following up, GCC has the [-Wc++-compat](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wc_002b_002b-compat) warning which can catch this in the future. Below is an excerpt of what it found from the code snippet in my previous comment. I also added the following line to see if it would catch the `this` keyword: `int this = 0;`

```
gpio.c:136:9: warning: identifier 'this' conflicts with C++ keyword [-Wc++-compat]
  136 |     int this = 0;
      |         ^~~~
gpio.c:142:16: warning: identifier 'public' conflicts with C++ keyword [-Wc++-compat]
  142 |     out_stream.public.putc(&out_stream.public, (int)'d');
      |                ^~~~~~
gpio.c:142:40: warning: identifier 'public' conflicts with C++ keyword [-Wc++-compat]
  142 |     out_stream.public.putc(&out_stream.public, (int)'d');
      |                                        ^~~~~~
```

If enabling this check across all of nuttx would be too restrictive (because of other, unwanted things it warns about) then an alternative would be a shellscript which compiles nuttx with this warning enabled and greps the output for "conflicts with C++ keyword". If found then the script would exit unsuccessfully. This could be a companion to `nxstyle`.

xiaoxiang781216, 2023-09-10:
> > it's wrong to use public(C++ keyword) in stream interface, please provide a patch to rename public to other word(e.g. common).
> 
> Should I also provide a patch for the `this` keyword being used?

Yes.

> Also would forbidding the use of C++ keywords in C be something to add to the [C coding standard](https://nuttx.apache.org/docs/latest/contributing/coding_style.html) as well as automated checks in `nxstyle`?

it will be great.

> Not sure if this is something that should be brought up in the mailing list or not.
> 

it's a natural requirement that all c header files should be work with c++ compiler.

> > 4. rtt serial console: [Add SEGGER RTT serial/console support #9777](https://github.com/apache/nuttx/pull/9777)
> 
> Thanks Xiang, it looks like this is equivalent to what I was looking for. I see that this PR introduced the `SERIAL_RTT` kconfig. This was after NuttX 12.2.1 was released so it didn't make the cut. For posterity, I'll post a code snippet of how I was able to get RTT working in my nuttx version in a C file:
> 
> ```
>   #include <nuttx/segger/rtt.h>
>   ....
> 
>   struct lib_rttoutstream_s out_stream = {0};
>  /* channel 0 is reserved by RTT for the Terminal*/
>   lib_rttoutstream_open(&out_stream, 0, CONFIG_SEGGER_RTT_BUFFER_SIZE_UP);
> 
>   /* the below should be equivalent to `out_stream.public.puts(&out_stream.public, "dan\r\n", sizeof("dan\r\n");`  */
>   out_stream.public.putc(&out_stream.public, (int)'d');
>   out_stream.public.putc(&out_stream.public, (int)'a');
>   out_stream.public.putc(&out_stream.public, (int)'n');
>   out_stream.public.putc(&out_stream.public, (int)'\r');
>   out_stream.public.putc(&out_stream.public, (int)'\n');
>   out_stream.public.putc(&out_stream.public, (int)'\0');
> 
>   while (1){
>         usleep(500);
>     }
> ```
> 

The code snippet is more like rtt syslog channel:
https://github.com/apache/nuttx/blob/master/drivers/segger/syslog_rtt.c
not rtt serial device:
https://github.com/apache/nuttx/blob/master/drivers/segger/serial_rtt.c

> Using Segger's [JLinkRTTViewer](https://www.segger.com/products/debug-probes/j-link/tools/rtt-viewer/), I attached to the existing J-link debug session I had with Segger Ozone and was able to see my log in Terminal 0: [image]

If you just want to see the log, syslog_rtt is better than serial_rtt. The later is designed to work with nsh through JLink.

danielappiagyei-bc, 2023-09-10:
**RE: replacing the usage of  C++ keywords from C headers**
Replacing even just `this` and `public` from https://github.com/apache/nuttx/blob/master/include/nuttx/streams.h will be a big change that touches a lot of files. That header defines a bunch of `stream` struct types that have a field named `public` (e.g., [here](https://github.com/apache/nuttx/blob/master/include/nuttx/streams.h#L140)). Therefore, any C source or header that includes stream.h and uses stream struct objects will need to be updated to use the new field name as well. I don't mind coming up with a renaming patch and submitting a PR, but I want to make sure that you all will be okay with such a big change beforehand.

Here are just a few examples I got when I:
- added the `-Wc++-compat` flag to `CFLAGS` in [IMXRT's Make.defs](https://github.com/apache/nuttx/blob/master/boards/arm/imxrt/imxrt1064-evk/scripts/Make.defs#L43)
- added `CONFIG_STREAM_RTT=y` to [IMXRT's defconfig](https://github.com/apache/nuttx/blob/master/boards/arm/imxrt/imxrt1064-evk/configs/nsh/defconfig)
- and added `#include <nuttx/segger/rtt.h>` to [nsh.h in nuttx-apps](https://github.com/apache/nuttx-apps/blob/master/nshlib/nsh.h)

```
stdio/lib_vdprintf.c:60:51: warning: identifier 'public' conflicts with C++ keyword [-Wc++-compat]                                                                                                                 
   60 |   lib_bufferedoutstream(&outstream, &rawoutstream.public);                                                                                                                                                 
      |                                                   ^~~~~~                                                                                                                                                   
stdio/lib_vdprintf.c:61:33: warning: identifier 'public' conflicts with C++ keyword [-Wc++-compat]                                                                                                                 
   61 |   ret = lib_vsprintf(&outstream.public, fmt, ap);                                                                                                                                                          
      |                                 ^~~~~~                                                                                                                                                                     
In file included from /home/daniel.appiagyei/projects/dan-nuttx/nuttx/libs/libc/libc.h:39,                                                                                                                         
                 from stdio/lib_vdprintf.c:45:                                                                                                                                                                     
stdio/lib_vdprintf.c:62:31: warning: identifier 'public' conflicts with C++ keyword [-Wc++-compat]                                                                                                                 
   62 |   lib_stream_flush(&outstream.public);                                                                                                                                                                     
      |                               ^~~~~~                                                                                                                                                                       
stdio/lib_vdprintf.c:62:31: warning: identifier 'public' conflicts with C++ keyword [-Wc++-compat]                                                                                                                 
   62 |   lib_stream_flush(&outstream.public);                                                                                                                                                                     
      |                               ^~~~~~                                                                                                                                                                       
CC:  stdio/lib_vsscanf.c stdio/lib_vsscanf.c: In function 'vsscanf':                                                                                                                                               
stdio/lib_vsscanf.c:53:60: warning: identifier 'public' conflicts with C++ keyword [-Wc++-compat]                                                                                                                  
   53 |   n = lib_vscanf((FAR struct lib_instream_s *)&meminstream.public, NULL, 
```

Is such a big change something you'd all be okay with?

> The code snippet is more like rtt syslog channel

Got it, thanks!

xiaoxiang781216, 2023-09-10:
If we want to support c++ code to use stream interface, this is the correct way to fix the problem.

danielappiagyei-bc, 2023-09-10:
> If we want to support c++ code to use stream interface, this is the correct way to fix the problem, do you have better method than rename?

Not at all! I'm new here so I just wanted to make sure I wouldn't be overstepping or breaking any rules by making such a big change. :-) Thanks!

xiaoxiang781216, 2023-09-11:
> > do you have better method than rename?
> 
> Not at all! I'm new here so I just wanted to make sure I wouldn't be overstepping or breaking any rules by making such a big change. :-) Thanks!

The change is fine since there is no the logic change, but a bunch of rename.

closed 2023-09-16 by xiaoxiang781216