apache/nuttx #10594: RISC-V linker fail, relocation truncated to fit: R_RISCV_HI20

TaiJuWu, 2023-09-12:
When I try to compile nuttx hash: f39de6fd1a
config: rv-virt:smp64
tookchain: riscv64-unknown-elf-gcc (g2ee5e430018) 12.2.0
```
riscv64-unknown-elf-ld: warning: /home/ubuntu/nuttxSpace/nuttx/nuttx has a LOAD segment with RWX permissions
/opt/riscv/lib/gcc/riscv64-unknown-elf/12.2.0/libgcc.a(_clzsi2.o): in function `__clzdi2':
/home/ubuntu/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv64-unknown-elf/libgcc/../../.././gcc/libgcc/libgcc2.c:690:(.text+0x18): relocation truncated to fit: R_RISCV_HI20 against symbol `__clz_tab' defined in .rodata section in /opt/riscv/lib/gcc/riscv64-unknown-elf/12.2.0/libgcc.a(_clz.o)
```
Does anyone meet this problem?

xiaoxiang781216, 2023-09-12:
Where you toolchain come from? This is a known issue for the new toolchain which lead https://github.com/apache/nuttx/pull/10306 is revert.

TaiJuWu, 2023-09-12:
> Where you toolchain come from? This is a known issue for the new toolchain which lead #10306 was reverted. Could you spend time to fix it?

My toolchain version is 12.2.0  comes from https://github.com/riscv-collab/riscv-gnu-toolchain.
If you have detailed step I can try to fix #10306 but I don't have many knowledge about compiler. 
By the way my mac is M1, Im not sure it fit to rest.

xiaoxiang781216, 2023-09-12:
Since it's the same problem, you can fix the problem with your current environment. One the fix is merged, I will bring back #10306 to ensure no new break happen in the future.

no1wudi, 2023-09-12:
Maybe we should build riscv toolchain from souce to match our requirement, by default the prebuilt toolchain use `-mcmodel=medlow` which may provide better performance but limit the jump range, we use `-mcmodel=medany`, they are not compatible in some scenarios, refer to https://github.com/riscv-collab/riscv-gcc/issues/153

TaiJuWu, 2023-09-12:
I find a [post](https://forums.sifive.com/t/link-error-relocation-truncated-to-fit-r-riscv-jal-against-symbol/4163) illustrate this problem and try to follow their step.
Original linker script is 
```
SECTIONS
{
  . = 0x80000000;

  .text :
    {
      _stext = . ;
      *(.text)
      *(.text.*)
      *(.gnu.warning)
      *(.stub)
      *(.glue_7)
      *(.glue_7t)
      *(.jcr)

      /* C++ support:  The .init and .fini sections contain specific logic
       * to manage static constructors and destructors.
       */

      *(.gnu.linkonce.t.*)
      *(.init)             /* Old ABI */
      *(.fini)             /* Old ABI */
      _etext = . ;
    }

  .rodata :
    {
      _srodata = . ;
      *(.rodata)
      *(.rodata1)
      *(.rodata.*)
      *(.gnu.linkonce.r*)
      _erodata = . ;
    }
.....
```
and their address  show 
```
0x000000008000051e                __clzdi2
0x0000000080017f20                __clz_tab
```

After I modify linker script to 
```
SECTIONS
{
  . = 0x80000000;

  .text :
    {
      _stext = . ;
      *(.text.*)
      *(.gnu.warning)
      *(.stub)
      *(.glue_7)
      *(.glue_7t)
      *(.jcr)
      *(.text)
      
      /* C++ support:  The .init and .fini sections contain specific logic
       * to manage static constructors and destructors.
       */

      *(.gnu.linkonce.t.*)
      *(.init)             /* Old ABI */
      *(.fini)             /* Old ABI */
      _etext = . ;
    }

  .rodata :
    {
      _srodata = . ;
      *(.rodata)
      *(.rodata1)
      *(.rodata.*)
      *(.gnu.linkonce.r*)
      _erodata = . ;
    }
```
Their address are
```
0x0000000080017f1e                __clzdi2
0x0000000080017f60                __clz_tab
```
But still failed.

pkarashchenko, 2023-09-12:
Could you please modify the linker script to become
```
SECTIONS
{
  . = 0x80000000;

  .text :
    {
      _stext = . ;
      *(.text) // key
      *(.text.*)
      *(.gnu.warning)
      *(.stub)
      *(.glue_7)
      *(.glue_7t)
      *(.jcr)

      /* C++ support:  The .init and .fini sections contain specific logic
       * to manage static constructors and destructors.
       */

      *(.gnu.linkonce.t.*)
      *(.init)             /* Old ABI */
      *(.fini)             /* Old ABI */
      _etext = . ;
    }

  .rodata :
    {
      _srodata = . ;
      *(.rodata)
      *(.rodata1)
      *(.rodata.*)
      *(.srodata)
      *(.srodata.*)
      *(.gnu.linkonce.r*)
      _erodata = . ;
    }
```
also it would be good to add `*(.sdata .sdata.* .sdata2.*)` to `.data` section I think in some place before `*(.gnu.linkonce.s.*)` should be fine and `*(.sbss*)` to `.bss` section.
Could you please try it out and feedback if that helps?

pkarashchenko, 2023-09-12:
I think you can try to follow https://github.com/apache/nuttx/blob/master/boards/risc-v/c906/smartl-c906/scripts/ld.script as an example

TaiJuWu, 2023-09-13:
> rv-virt:smp64

It still failed.

TaiJuWu, 2023-09-13:
> Could you please modify the linker script to become
>
> ```
> SECTIONS
> {
>   . = 0x80000000;
>
>   .text :
>     {
>       _stext = . ;
>       *(.text) // key
>       *(.text.*)
>       *(.gnu.warning) [...]

It still failed.

pkarashchenko, 2023-09-15:
I downloaded and installed the same toolchain and am able to reproduce an issue. I think the case is that toolchain is build without support of `medany` model as `riscv64-unknown-elf-gcc -v` outputs
```
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/usr/share/riscv64-elf-gcc-nightly-2023.09.13/bin/../libexec/gcc/riscv64-unknown-elf/12.2.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /home/runner/work/riscv-gnu-toolchain/riscv-gnu-toolchain/gcc/configure --target=riscv64-unknown-elf --prefix=/opt/riscv --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion= --with-system-zlib --enable-tls --with-newlib --with-sysroot=/opt/riscv/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=.././gcc --disable-multilib --with-abi=lp64d --with-arch=rv64gc --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-Os    -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os    -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 12.2.0 ()
```
I think the best would be to use toolchain from https://github.com/apache/nuttx/pull/10306

TaiJuWu, 2023-09-15:
I can update some information
I still use the same toolchain but I use tag: nuttx-12.2.1 and it can work successfully.

pkarashchenko, 2023-09-15:
That is interesting. Could you please share map files with and without the issue?

TaiJuWu, 2023-09-15:
> That is interesting. Could you please share map files with and without the issue?

Sorry, I forgot I using arm toolchain to work instead of riscv.

pkarashchenko, 2023-09-16:
I recommend to use a proper toolchain and close this issue. Basically we need toolchain to support `-mcmodel=medany` that https://github.com/riscv-collab/riscv-gnu-toolchain obviously does not.

TaiJuWu, 2023-09-16:
> I recommend to use a proper toolchain and close this issue. Basically we need toolchain to support `-mcmodel=medany` that https://github.com/riscv-collab/riscv-gnu-toolchain obviously does not.

I got it. Thanks your reply.

pkarashchenko, 2023-09-16:
If you do not need to compile C++ code then you can use the latest toolchain from https://github.com/sifive/freedom-tools/releases

xiaoxiang781216, 2023-09-16:
@pkarashchenko do you find the latest prebuilt toolchain from the well-known provider which support -mcmodel=medany?

pkarashchenko, 2023-09-16:
The issue here is that SiFive stopped the creation of prebuilt versions (not sure why, but still this is the fact) and all other places I found seem to be tailored to a specific use case.

xiaoxiang781216, 2023-09-16:
Yes, we need find a provider which provide the toolchain with  -mcmodel=medany, otherwise we can't:

1. Upgrade libcxx to the new version
2. Fix macOS ci error

Both are important and urgent to us.

xiaoxiang781216, 2023-09-18:
toolchain upgrade try before: https://github.com/apache/nuttx/pull/10306 and https://github.com/apache/nuttx/pull/10348.

lupyuen, 2023-09-18:
Hi All: Wonder if the xPack GNU RISC-V Embedded GCC might fix this problem?

https://xpack.github.io/dev-tools/riscv-none-elf-gcc/install/

I built NuttX for JH7110 (RISCV64) with the xPack Toolchain, and it boots OK on Star64 SBC. I haven't tested Math Functions yet:

https://lupyuen.github.io/articles/riscv#appendix-xpack-gnu-risc-v-embedded-gcc-toolchain-for-64-bit-risc-v

I'm exploring xPack as an alternative toolchain, since SiFive Freedom Tools won't run on Arm64 Linux (Raspberry Pi OS). I tried gcc-riscv64-unknown-elf on Arm64 Debian, but it doesn't include "math.h":

https://lupyuen.github.io/articles/release#appendix-missing-mathh

no1wudi, 2023-09-18:
xPack's toolchain may not fix this problem, I got its compile configurations by `--verbose`:

```
riscv-none-elf-gcc --verbose
Using built-in specs.
COLLECT_GCC=./riscv-none-elf-gcc
COLLECT_LTO_WRAPPER=/workspaces/xpack-riscv-none-elf-gcc-13.2.0-1/bin/../libexec/gcc/riscv-none-elf/13.2.0/lto-wrapper
Target: riscv-none-elf
Configured with: /__w/riscv-none-elf-gcc-xpack/riscv-none-elf-gcc-xpack/build/linux-x64/sources/gcc-13.2.0/configure --prefix=/__w/riscv-none-elf-gcc-xpack/riscv-none-elf-gcc-xpack/build/linux-x64/application --with-sysroot=/__w/riscv-none-elf-gcc-xpack/riscv-none-elf-gcc-xpack/build/linux-x64/application/riscv-none-elf --with-native-system-header-dir=/include --infodir=/__w/riscv-none-elf-gcc-xpack/riscv-none-elf-gcc-xpack/build/linux-x64/x86_64-pc-linux-gnu/install/share/info --mandir=/__w/riscv-none-elf-gcc-xpack/riscv-none-elf-gcc-xpack/build/linux-x64/x86_64-pc-linux-gnu/install/share/man --htmldir=/__w/riscv-none-elf-gcc-xpack/riscv-none-elf-gcc-xpack/build/linux-x64/x86_64-pc-linux-gnu/install/share/html --pdfdir=/__w/riscv-none-elf-gcc-xpack/riscv-none-elf-gcc-xpack/build/linux-x64/x86_64-pc-linux-gnu/install/share/pdf --build=x86_64-pc-linux-gnu --host=x86_64-pc-linux-gnu --target=riscv-none-elf --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libsanitizer --disable-libssp --disable-nls --disable-shared --disable-threads --disable-tls --enable-checking=release --enable-languages=c,c++,fortran --with-gmp=/__w/riscv-none-elf-gcc-xpack/riscv-none-elf-gcc-xpack/build/linux-x64/x86_64-pc-linux-gnu/install --with-newlib --with-pkgversion='xPack GNU RISC-V Embedded GCC x86_64' --with-gnu-as --with-gnu-ld --with-system-zlib --with-abi=ilp32 --with-arch=rv32imac --enable-multilib
Thread model: single
Supported LTO compression algorithms: zlib zstd
gcc version 13.2.0 (xPack GNU RISC-V Embedded GCC x86_64) 
```
The toolchain from Ubuntu/Debian may works since it compiled with `-mcmodel=medany` but it lacks of libm:
```
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/riscv64-unknown-elf/9.3.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: ../configure --build=x86_64-linux-gnu --prefix=/usr --includedir='/usr/include' --mandir='/usr/share/man' --infodir='/usr/share/info' --sysconfdir=/etc --localstatedir=/var --disable-silent-rules --libdir='/usr/lib/x86_64-linux-gnu' --libexecdir='/usr/lib/x86_64-linux-gnu' --disable-maintainer-mode --disable-dependency-tracking --target=riscv64-unknown-elf --prefix=/usr --infodir=/usr/share/doc/gcc-riscv64-unknown-elf/info --mandir=/usr/share/man --htmldir=/usr/share/doc/gcc-riscv64-unknown-elf/html --pdfdir=/usr/share/doc/gcc-riscv64-unknown-elf/pdf --bindir=/usr/bin --libexecdir=/usr/lib --libdir=/usr/lib --with-pkgversion= --disable-shared --disable-threads --enable-languages=c,c++ --enable-tls --with-newlib --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --with-system-zlib --enable-checking=yes --enable-multilib --with-abi=lp64d --disable-libstdcxx-pch --disable-libstdcxx --disable-fixinc --with-arch=rv64imafdc --with-gnu-as --with-gnu-ld --with-as=/usr/lib/riscv64-unknown-elf/bin/as --with-ld=/usr/lib/riscv64-unknown-elf/bin/ld AR_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/ar AS_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/as NM_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/nm LD_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/ld OBJDUMP_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/objdump RANLIB_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/ranlib READELF_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/readelf STRIP_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/strip CFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3seJsn/gcc-riscv64-unknown-elf-9.3.0=. -fstack-protector-strong' CPPFLAGS='-Wdate-time -D_FORTIFY_SOURCE=2' CXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3seJsn/gcc-riscv64-unknown-elf-9.3.0=. -fstack-protector-strong' FCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3seJsn/gcc-riscv64-unknown-elf-9.3.0=. -fstack-protector-strong' FFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3seJsn/gcc-riscv64-unknown-elf-9.3.0=. -fstack-protector-strong' GCJFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3seJsn/gcc-riscv64-unknown-elf-9.3.0=. -fstack-protector-strong' LDFLAGS='-Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-z,now' OBJCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3seJsn/gcc-riscv64-unknown-elf-9.3.0=. -fstack-protector-strong' OBJCXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3seJsn/gcc-riscv64-unknown-elf-9.3.0=. -fstack-protector-strong' 'CFLAGS_FOR_TARGET=-Os -mcmodel=medany' 'CXXFLAGS_FOR_TARGET=-Os -mcmodel=medany'
Thread model: single
gcc version 9.3.0 ()
```

xiaoxiang781216, 2023-09-18:
@lupyuen libm issue could be fixed by select from nuttx, newlib, libmcs and openlibm if toolchain doesn't provide one:
https://github.com/apache/nuttx/blob/master/libs/libm/Kconfig#L6-L61

lupyuen, 2023-09-18:
@no1wudi I found this message under xPack v12.1.0.2, does it mean that `-mcmodel=medany` is already selected?

https://xpack.github.io/blog/2022/05/18/riscv-none-elf-gcc-v12-1-0-2-released/#-mcmodelmedany

> The libraries are compiled with `-O2 -mcmodel=medany`. The nano version is compiled with `-Os -mcmodel=medany`.

> Important: It is mandatory for the applications to be compiled with `-mcmodel=medany`, otherwise the link might fail.

> [image]

It doesn't appear in the binary though:

```text
→ ./riscv-none-elf-gcc  --verbose
Using built-in specs.
COLLECT_GCC=./riscv-none-elf-gcc
COLLECT_LTO_WRAPPER=/private/tmp/xpack-riscv-none-elf-gcc-12.1.0-2/bin/../libexec/gcc/riscv-none-elf/12.1.0/lto-wrapper
Target: riscv-none-elf
Configured with: /Users/ilg/Work/riscv-none-elf-gcc-12.1.0-2/darwin-x64/sources/gcc-12.1.0/configure
  --prefix=/Users/ilg/Work/riscv-none-elf-gcc-12.1.0-2/darwin-x64/install/riscv-none-elf-gcc
  --prefix=/Users/ilg/Work/riscv-none-elf-gcc-12.1.0-2/darwin-x64/install/riscv-none-elf-gcc
  --infodir=/Users/ilg/Work/riscv-none-elf-gcc-12.1.0-2/darwin-x64/install/riscv-none-elf-gcc/share/doc/info
  --mandir=/Users/ilg/Work/riscv-none-elf-gcc-12.1.0-2/darwin-x64/install/riscv-none-elf-gcc/share/doc/man
  --htmldir=/Users/ilg/Work/riscv-none-elf-gcc-12.1.0-2/darwin-x64/install/riscv-none-elf-gcc/share/doc/html
  --pdfdir=/Users/ilg/Work/riscv-none-elf-gcc-12.1.0-2/darwin-x64/install/riscv-none-elf-gcc/share/doc/pdf
  --build=x86_64-apple-darwin17.7.0
  --host=x86_64-apple-darwin17.7.0
  --target=riscv-none-elf
  --disable-nls
  --disable-shared
  --disable-threads
  --disable-tls
  --enable-checking=release
  --enable-languages=c,c++,fortran
  --with-gmp=/Users/ilg/Work/riscv-none-elf-gcc-12.1.0-2/darwin-x64/install/libs
  --with-newlib
  --with-pkgversion='xPack GNU RISC-V Embedded GCC x86_64'
  --with-system-zlib
  --with-abi=ilp32
  --with-arch=rv32imac
  --enable-multilib
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 12.1.0 (xPack GNU RISC-V Embedded GCC x86_64)
```

no1wudi, 2023-09-18:
> @no1wudi I found this message under xPack v12.1.0.2, does it mean that `-mcmodel=medany` is already selected? [...]

Thanks for your notifacation, maybe we can consider to using xPack's toolchain for risc-v in CI.

closed 2023-09-18 by xiaoxiang781216