apache/nuttx #10642: nuttx didn't use atomic instruction?

TaiJuWu, 2023-09-15:
Many architecture offer atomic instruction like compare and swap, test and set....
But I didn't find any atomic operation in nuttx kernel except atomic testing.

Is there any reason so we don't use them?
Because recently I want to optimize fair spin lock with lock-free skill.
But the basic requirements are atomic instruction.

xiaoxiang781216, 2023-09-16:
atomic operation is defined since c11(https://en.cppreference.com/w/c/language/atomic), and the implementation is provided by toolchain already(e.g, libgcc.a from gcc). That's why you can't find the implementation from nuttx, an atmic test was added recently to help you check the atomic support with your toolchain:
https://github.com/apache/nuttx-apps/tree/master/testing/atomic.
For the old compiler, nuttx provide a simple implementation which ensure the atomic by disabling interrupt temporarily:
https://github.com/apache/nuttx/blob/master/libs/libc/machine/arch_atomic.c
Of cause, this implementation just works with one CPU case.

TaiJuWu, 2023-09-16:
> atomic operation is defined since c11(https://en.cppreference.com/w/c/language/atomic), and the implementation is provided by toolchain already(e.g, libgcc.a from gcc). That's why you can't find the [...]

https://github.com/apache/nuttx/blob/master/libs/libc/machine/arch_atomic.c
It seems to work on multi-core.
The share variable already is protected by spin lock and disable interrupt.
Why did you say only work on one CPU?

mu578, 2023-09-16:
> > atomic operation is defined since c11(https://en.cppreference.com/w/c/language/atomic), and the implementation is provided by toolchain already(e.g, libgcc.a from gcc). That's why you can't find the implementation from nuttx, an atmic test was added recently to help you check the atomic support with your toolchain:
> > https://github.com/apache/nuttx-apps/tree/master/testing/atomic.
> > For the old compiler, nuttx provide a simple implementation which ensure the atomic by disabling interrupt temporarily:
> > https://github.com/apache/nuttx/blob/master/libs/libc/machine/arch_atomic.c
> > Of cause, this implementation just works with one CPU case.
> 
> First thank you for your reply.
> 
> https://github.com/apache/nuttx/blob/master/libs/libc/machine/arch_atomic.c It seems to work on multi-core. The share variable already is protected by spin lock and disable interrupt. Why did you say only work on one CPU?

yes here atomic are builtin with a spin lock ... where you want to build your spin lock relaying on atomic ops ; gcc and clang have builtin atomic ops ehehe,

however for gcc not prior 4.7, have a look here, that's pasta dig up

https://gist.github.com/mu578/b7bd9ebd2131c02d73d25f1a6f927751

xiaoxiang781216, 2023-09-17:
> https://github.com/apache/nuttx/blob/master/libs/libc/machine/arch_atomic.c It seems to work on multi-core. The share variable already is protected by spin lock and disable interrupt. Why did you say only work on one CPU?

Since the code can just disable the current core interrupt, not other cores. Even it can disable other cores, it doesn't work as your expect, because other cores are still running and may try to acquire the same lock.

TaiJuWu, 2023-09-17:
> Since the code can just disable the current core interrupt, not other cores. Even it can disable other cores, it doesn't work as your expect, because other cores are still running and may try to [...]

Yes, It acquire same lock and spin here.
I guess that even if disable core interrupt but the shared variable maybe happen in global interrupt context.
So the library only works on single core.

closed 2023-09-17 by TaiJuWu