apache/nuttx #10754: qume performance measure

TaiJuWu, 2023-09-21:
I want to test my code for nuttx.
Is it available test on QEMU?

I attempted use os perf but pipe2 does not work for qemu-armv8a:nsh_smp
Is it resolved? Or I must measure performance on real hardware?

xiaoxiang781216, 2023-09-21:
> I want to test my code for nuttx. Is it available test on QEMU?
> 

Yes.

> I attempted use os perf but pipe2 does not work for qemu-armv8a:nsh_smp Is it resolved?

what' problem?

> Or I must measure performance on real hardware?
> 

it's always better to meaure on the real hardware.

> The osperf always trigger memory dump and I don't find any documentation about it. Can anyone give me some suggestion?

@Gary-Hobson could you take a look?

TaiJuWu, 2023-09-22:
> Yes.
>
>
> what' problem?
>
>
> it's always better to meaure on the real hardware.
>
>
> @Gary-Hobson could you take a look?

When I use QEMU to measure performance the time looks incorrect, even only pass 1s on real world but the result shows 14s.

 So I’m not sure this behavior is available.

Gary-Hobson, 2023-09-22:
> > > I want to test my code for nuttx. Is it available test on QEMU?
> > 
> > 
> > Yes.
> > > I attempted use os perf but pipe2 does not work for qemu-armv8a:nsh_smp Is it resolved?
> > 
> > 
> > what' problem?
> > > Or I must measure performance on real hardware?
> > 
> > 
> > it's always better to meaure on the real hardware.
> > > The osperf always trigger memory dump and I don't find any documentation about it. Can anyone give me some suggestion?
> > 
> > 
> > @Gary-Hobson could you take a look?
> 
> When I use QEMU to measure performance the time looks incorrect, even only pass 1s on real world but the result shows 14s.
> 
> So I’m not sure this behavior is available and I have attempted to find the answer but failed.

There were some issues running on arm64 which I have fixed
I also found that the time you mentioned is incorrect. I am analyzing the reasons now and may have results later.

[image]

TaiJuWu, 2023-09-22:
> There were some issues running on arm64 which I have fixed I also found that the time you mentioned is incorrect. I am analyzing the reasons now and may have results later.
>
> [image]

Thanks a lot.

Gary-Hobson, 2023-09-22:
@TaiJuWu You can use this PR(https://github.com/apache/nuttx-apps/pull/2085) to resolve runtime errors.

The time display error is due to the fact that in SMP mode, the time obtained by different CPUs through up_perf_gettime is not synchronized, and end may be smaller than start, resulting in time errors.

[image]

And you need to enable the following configuration
```
CONFIG_DEBUG_SCHED_INFO=n
CONFIG_PIPES=y
CONFIG_TESTING_OSPERF=y
```

closed 2023-09-23 by xiaoxiang781216