apache/nuttx #10794: Menuconfig failing after #10763 in Manjaro's zsh

tmedicci, 2023-09-24:
After https://github.com/apache/nuttx/pull/10763, `make menuconfig` throws the following error being executed on Manjaro's zsh:

```
CP: arch/dummy/Kconfig to /home/tiago/Documents/tiago/learning/nuttx/nuttxspace/nuttx/arch/dummy/dummy_kconfig
CP: boards/dummy/Kconfig to /home/tiago/Documents/tiago/learning/nuttx/nuttxspace/nuttx/boards/dummy/dummy_kconfig
LN: platform/board to /home/tiago/Documents/tiago/learning/nuttx/nuttxspace/apps/platform/dummy
LN: include/arch to arch/sim/include
LN: include/arch/board to /home/tiago/Documents/tiago/learning/nuttx/nuttxspace/nuttx/boards/sim/sim/sim/include
LN: drivers/platform to /home/tiago/Documents/tiago/learning/nuttx/nuttxspace/nuttx/drivers/dummy
LN: include/arch/chip to /home/tiago/Documents/tiago/learning/nuttx/nuttxspace/nuttx/arch/sim/include/sim
LN: arch/sim/src/chip to /home/tiago/Documents/tiago/learning/nuttx/nuttxspace/nuttx/arch/sim/src/sim
LN: arch/sim/src/board to /home/tiago/Documents/tiago/learning/nuttx/nuttxspace/nuttx/boards/sim/sim/sim/src
Traceback (most recent call last):
  File "/usr/lib/python3.11/curses/__init__.py", line 78, in wrapper
    cbreak()
_curses.error: cbreak() returned ERR

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/tiago/.local/bin/menuconfig", line 8, in <module>
    sys.exit(_main())
             ^^^^^^^
  File "/usr/lib/python3.11/site-packages/menuconfig.py", line 663, in _main
    menuconfig(standard_kconfig(__doc__))
  File "/usr/lib/python3.11/site-packages/menuconfig.py", line 732, in menuconfig
    print(curses.wrapper(_menuconfig))
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/curses/__init__.py", line 100, in wrapper
    nocbreak()
_curses.error: nocbreak() returned ERR
make: *** [tools/Unix.mk:685: menuconfig] Error 1
```

This seems to be valid for any defconfig. From a clean repository, it can be easily reproduced using `make -j distclean && ./tools/configure.sh sim:nsh && make menuconfig`

acassis, 2023-09-24:
@tmedicci this error happen when a curses application is not running in real terminal:

https://stackoverflow.com/questions/52747655/why-do-i-get-curses-error-cbreak-returned-err-when-using-tensorflow-cli-de

https://github.com/ulfalizer/Kconfiglib/issues/64

xuxin930, 2023-09-25:
hi @tmedicci @acassis
the warning check has been completed before the menuconfig execution phase, so lets ignored here.
I have already submitted a fix patch, please review https://github.com/apache/nuttx/pull/10801

closed 2023-09-25 by xiaoxiang781216