apache/nuttx #10821: Board Initialization: Architecture-specific initialization vs Custom board late initialization

Rdk-T, 2023-09-27:
When putting an application on ENTRYPOINT instead of launching it manually through nsh, some driver initialization is missing.

This is explained in the faq. https://nuttx.apache.org/docs/latest/faq/index.html#why-after-putting-my-application-on-entrypoint-it-stops-to-work

Is there a drawback to replace `CONFIG_NSH_ARCHINIT=y` by `CONFIG_BOARD_LATE_INITIALIZE=y` in the boards defconfig ? 
Is `CONFIG_NSH_ARCHINIT` useful compared to `CONFIG_BOARD_LATE_INITIALIZE` ?

This way new users won't have to figure out why their application works when launching from nsh and doesn't when launching as ENTRYPOINT.

acassis, 2023-09-29:
@Rdk-T I think initially CONFIG_NSH_ARCHINIT was the only option to run some board code initialization, later on CONFIG_BOARD_LATE_INITIALIZE and also early board initialization were included. I think eventually we will get rid of CONFIG_NSH_ARCHINIT to have only generic initialization that applies to all applications. No specially citizen app!

acassis, 2023-10-02:
@xiaoxiang781216 @patacongo @masayuki2009 what you think? Does it make sense to simplify the initialization process on NuttX to avoid having CONFIG_NSH_ARCHINIT ?

I think we need a better documentation about it as well, in the past I documented it in my blog with comments that Greg posted in the mailing list:

https://acassis.wordpress.com/2015/07/01/nuttx-board-initialization-where-can-i-put-my-code-to-be-execute/

https://acassis.wordpress.com/2014/09/20/how-to-do-late-board-initialization-on-nuttx/

But of course, use a generic solution to applies to Apps and NSH will be better