apache/nuttx #10853: imxrt/lpi2c multibyte read broken following DMA addition

thebolt, 2023-10-04:
We are on Nuttx 12.2.1 release, with and without commit afdce6e8c2d9557c9032c91aebc997ba555bb721 back-applied.
The device is a iMXRT 1062, and communicating with an RX8900 RTC using an in-house driver.

Upon upgrading from 10.0 series to 12.2, I have discovered that the lpi2c driver for imxrt has been broken. Plain 12.2.1 I have not yet got any I2C trace, but following very weird issues with it I applied commit afdce which mostly seems to resolve the issues with one exception.

A sequence of write (1 byte) followed by read (7 bytes) fails, with the following I2C trace.

```
 1. STATUS: 00000001 COUNT:   1 EVENT: START/RESTART  ( 5) PARM: 00000002 TIME: 1
 2. STATUS: 00000000 COUNT:   0 EVENT: SENDBYTE       ( 2) PARM: 00000001 TIME: -5
 3. STATUS: 03000001 COUNT:   1 EVENT: START/RESTART  ( 5) PARM: 00000001 TIME: 1
 4. STATUS: 03000103 COUNT:   1 EVENT: RCVBYTE        ( 3) PARM: 00000007 TIME: 1
```

After that the _transfer function returns.

My initial analysis is that the check on line 1597  (https://github.com/apache/nuttx/blob/071fb18501a4198b21693bc089142039bb9dc1be/arch/arm/src/imxrt/imxrt_lpi2c.c#L1597) is the issue. When the first rcvbyte isr is received we did just do a start/restart, so EPF is set, however it does not mean the _current_ message is over, only that the _last_ message was over.

acassis, 2023-10-04:
@davids5 git blame pointed you, could you please take a look? :-)

acassis, 2023-10-04:
@thebolt please consider integrating RX8900 RTC into mainline, it will help more people.

davids5, 2023-10-05:
@acassis I will when I can get to it.... It will be a while.

thebolt, 2023-10-11:
Upstreaming that driver (and other local changes) is on my list, however at the moment a bit constrained by internal budget.

If the issue cannot be resolved timely (and now there will be another release with the issue), maybe you should consider reverting the commits breaking it? It was working before, now it is fully broken.

acassis, 2023-10-12:
Yes, normally we revert as last option when nobody is able to fix the issue. I think @davids5 will be find it!

acassis, 2023-10-15:
@thebolt maybe it is related with this memory issue reported here: https://github.com/apache/nuttx/issues/10912

thebolt, 2023-10-16:
It's most certainly not related. 

First of all I'm not using DMA for I2C, we are still using interrupt driven transfers (for an RTC its more than enough).

Secondly, I think (but haven't been able to verify) that the problem is a change in logic of stop handling. The logic as it is now will detect a "stop" condition whenever the bit is set, however it does not take into account that a restart also sets the stop bit, meaning that after the first byte after a restart it will just end the transfer.. this was not the case before, then it would only look at the end of transfer condition if the internal message buffers were empty.

davids5, 2023-10-16:
@thebolt Can you supply the data and flag setting that are failing along with the defconfig?

thebolt, 2023-10-16:
Sure. The defconfig is attached, and below is a code excerpt with the setup of the I2C message. It's part of our RTC driver which is not 100% ready to be publicly released, but the relevant part is:

```
#define CONFIG_RX8900_I2C_FREQUENCY 400000

#define RX8900_I2C_ADDRESS       0x32
#define RX8900_I2C_TRY_COUNT     4

#define RX8900_SEC               0x00

static int rx8900_read_regs(FAR struct rx8900_rtc_lowerhalf_s *priv,
                            uint8_t reg,
                            uint8_t count,
                            uint8_t *values)
{
    struct i2c_msg_s msg[2] = { 0 };
    int try = RX8900_I2C_TRY_COUNT;
    int ret;

    msg[0].frequency = CONFIG_RX8900_I2C_FREQUENCY;
    msg[0].addr = RX8900_I2C_ADDRESS;
    msg[0].flags = I2C_M_NOSTOP;
    msg[0].buffer = &reg;
    msg[0].length = 1;

    msg[1].frequency = CONFIG_RX8900_I2C_FREQUENCY;
    msg[1].addr = RX8900_I2C_ADDRESS;
    msg[1].flags = I2C_M_READ;
    msg[1].buffer = values;
    msg[1].length = count;

    /*
     * There is a 61µs window during which the RTC does not acknowledge I2C
     * transfers. In that case, ensure that there are multiple attempts.
     */
    do {
        ret = I2C_TRANSFER(priv->i2c, msg, 2);
    } while ((ret == -ENXIO || ret == -EIO) && --try);

    if (ret < 0) {
        rtcerr("ERROR: Failed to read registers: 0x%02x..0x%02x (err %d)\n", reg, reg + count, ret);
        return ret;
    }

    return OK;
}

void cat(void)
{
    ...
    int ret = rx8900_read_regs(priv, RX8900_SEC, 7, date);
    ...
}
```
[defconfig.txt](https://github.com/apache/nuttx/files/12915068/defconfig.txt)

davids5, 2023-10-20:
@thebolt I did find an issue with the wait time being 0 with CONFIG_xxxx_LPI2C_DYNTIMEO being used on the s32k1. 
but I see you are not using that. But do Check CONFIG_xxx_LPI2CTIMEOTICKS value.

I need to test this on HW I have with the same config.  I will need to cobble something together I am running behind, but looking into it.

davids5, 2023-10-23:
@thebolt would you please test this patch

[0006-imxrt-lpi2c-end-only-on-stop-with-end-of-packet.patch](https://github.com/apache/nuttx/files/13072552/0006-imxrt-lpi2c-end-only-on-stop-with-end-of-packet.patch)

thebolt, 2023-10-24:
Thanks for the fix. I will try to test it over the next few days (working on another product atm and takes a bit to switch over the lab setup); I'll let you know and close this once I verified it.

closed 2023-10-24 by pkarashchenko

davids5, 2023-10-25:
@thebolt - There was more see https://github.com/apache/nuttx/pull/11033