apache/nuttx #10906: Github README Improvement

fdcavalcanti, 2023-10-13:
Hi everyone!

I would like to propose a new README page on Github. The current one is a big wall of text which is already covered in the documentation.
I think we can do something smaller and cleaner just by following some examples online.
Sure, it's a visual thing but I personally think this can help attract more people and gives a "cleaner" look!

Here's what I propose, following these repos as reference: [link 1](https://github.com/zephyrproject-rtos/zephyr), [link 2](https://github.com/pytest-dev/pytest), [link 3](https://github.com/numpy/numpy)

- Put the NuttX logo on top
- Add badges showing status of build, docs, version, etc
- Quick intro to what NuttX is
- Topics on contributing, documentation and getting started

I would love to hear some more ideas from the community, please feel free to add them here.

Filipe

acassis, 2023-10-13:
I think this is a good suggestion!

fdcavalcanti, 2023-10-17:
I've created a sample on my fork: [link](https://github.com/fdcavalcanti/nuttx/tree/new-readme)
We can add more badges, improve the logo, wherever is necessary.

fdcavalcanti, 2023-10-19:
See #10980

closed 2023-10-28 by xiaoxiang781216