apache/nuttx #10931: Thread visibility in the NuttX simulator

lulingar, 2023-10-17:
Up until now, I have been relying on `sim:vpnkit` (the simulator) for developing and testing my software before I use a real hardware target. However, AFAIK it is not possible to have visibility of threads during debugging with the sim target. Is this correct? 

I'm considering different simulation targets, such as x86(-64) or the one used in https://nuttx.apache.org/docs/latest/guides/qemugdb.html, in order to have thread visibility during debugging. What do you recommend?

xiaoxiang781216, 2023-10-17:
you have two methods in case of simulator:

1. launch gdb as normal with https://github.com/apache/nuttx/tree/master/tools/gdb which work with simulator too
2. Enable https://github.com/apache/nuttx-apps/tree/master/system/gdbstub and attach gdb to it as remote debug

The second method can't set break pointer yet, utils @anjiahao1 upstream kgdbstub version.
For real hardware, you can try https://github.com/apache/nuttx/blob/master/tools/jlink-nuttx.c.

lulingar, 2023-10-17:
Thank you @xiaoxiang781216. Are you aware of any major pitfall with regard to behavior of read/write/poll operations of pipes among threads in the simulator? I ask because, the application I'm working on, shows no issues with this when built as a linux program, but when built into the nuttx simulator, it seems to block.

xiaoxiang781216, 2023-10-18:
> Thank you @xiaoxiang781216. Are you aware of any major pitfall with regard to behavior of read/write/poll operations on pipes among threads in the simulator? I ask because, the application I'm working on, shows no issues with this when built as a linux program, but when built into the nuttx simulator, it seems to block.

poll work well very with both simulator and real device; we are building a very complex framework which use poll heavily, so poll implementation is solid in general. Recently, @thebolt report a special corner case which is fixed by: https://github.com/apache/nuttx/pull/10706