apache/nuttx #10967: Does this project belong to Xiaomi Company? 这个项目归属于小米公司么？

david1025, 2023-10-18:
Many people say that Nuttx belongs to Xiaomi and was developed by Xiaomi themselves. Do I want to confirm if this is true?
非常多的人说nuttx现在归属于小米，是小米自研的，我想确认下这个是真的么？

Is it true that some people say the founder works at Xiaomi?
还有人说创始人在小米工作，是真的么？

I hope someone can clarify it

[image]

xiaoxiang781216, 2023-10-18:
> Many people say that Nuttx belongs to Xiaomi and was developed by Xiaomi themselves. Do I want to confirm if this is true? 非常多的人说nuttx现在归属于小米，是小米自研的，我想确认下这个是真的么？
> 

NuttX is an open-source project and donate to Apache foundation in 2019 by Gregory Nutt. The ownership is unambiguous:
https://nuttx.apache.org/

Since NuttX is a general RTOS kernel, very like Linux, there are still many works for various product (e.g., IoT module, Smart speaker, Band/Watch...). This additional work is part of Xiaomi Vela, here is a Vela introduction video shared on NuttX Online Workshop 2021:
https://www.youtube.com/watch?v=P-QicZgX7Jo
In summary, Vela+NuttX is very like Android+Linux.

> Is it true that some people say the founder works at Xiaomi? 还有人说创始人在小米工作，是真的么？
> 

Gregory NuttX isn't a Xiaomi employee, but a Xiaomi technical advisor from 2019.

xiaoxiang781216, 2023-10-18:
let's close it, @david1025 if you have more question, please open again.

closed 2023-10-18 by xiaoxiang781216