apache/nuttx #10977: connect() fails with EACCES error when CONFIG_NET_USRSOCK is enabled.

SPRESENSE, 2023-10-19:
It seems that there is a bug that causes connect() to fail with an EACCES error when CONFIG_NET_USRSOCK is enabled.
This bug appears to have been introduced in the PR of https://github.com/apache/nuttx/pull/7291
When a UDP socket is created and connect() is executed, between usrsock_setup_request_callback() and do_connect_request(),
the USRSOCK_EVENT_SENDTO_READY event is sent by the usock daemon for that created socket, it seems that do_connect_request() will return EACCES.
Since a UDP socket can be sent anytime upon creation, it is a regular case that a USRSOCK_EVENT_SENDTO_READY event is sent at this time.
Therefore, I think the problem is that conncet() causes an error when the USRSOCK_EVENT_SENDTO_READY event is sent during the above gap.
Could you please consider a countermeasure?

xiaoxiang781216, 2023-10-19:
@zhhyu7 could you take look?

zhhyu7, 2023-10-19:
> @zhhyu7 could you take look?

ok, let me take a look.

zhhyu7, 2023-10-19:
@SPRESENSE Which driver is used to read the usrsock request event in this scenario?  when the USRSOCK_REQUEST_SOCKET case, please try to combine the response with the event like https://github.com/apache/nuttx/blob/master/drivers/usrsock/usrsock_rpmsg_server.c#L300