apache/nuttx #10981: Question about enter_critical_section

TaiJuWu, 2023-10-19:
I have three questions about  enter_critical_section.

First.
When I wrote spinlock code, I found spinlock_irqsave can disable interrupt.
Why we need to use enter_critical_section in [mm_alloc](https://github.com/apache/nuttx/blob/a521fd7b82dec43871b4f56a6ae0d85d124158e9/mm/mm_heap/mm_malloc.c#L50) instead of spinlock_irqsave?
The operation is very light.
And I have read https://github.com/apache/nuttx/issues/2213#issuecomment-731052528, enter_critical_section may make caller sleep and cause delay.

Second.
And there is a [archive](https://www.mail-archive.com/dev@nuttx.apache.org/msg04078.html) describe `the critical section is left automatically when the context switch occurs` but I can't find any code about this description.

Three.
The critical section is in oder to resolving some architecture can't disable global interrupt.
But why other RTOS projects don't have this feature?

xiaoxiang781216, 2023-10-20:
> I have some questions about enter_critical_section.
> 
> First. When I wrote spinlock code, I found spinlock_irqsave can disable interrupt. Why we need to use enter_critical_section in [mm_alloc](https://github.com/apache/nuttx/blob/a521fd7b82dec43871b4f56a6ae0d85d124158e9/mm/mm_heap/mm_malloc.c#L50) instead of spinlock_irqsave? The operation is very light. And I have read [#2213 (comment)](https://github.com/apache/nuttx/issues/2213#issuecomment-731052528), enter_critical_section may make caller sleep and cause delay.
> 

Yes, it should be replaced with spinlock. Actually, we have an issue track this: https://github.com/apache/nuttx/issues/1144.

> Second. And there is a [archive](https://www.mail-archive.com/dev@nuttx.apache.org/msg04078.html) describe `the critical section is left automatically when the context switch occurs` but I can't find any code about this description.
> 

Because the irq status is part of per thread register context, the irq status is automatically restored when arch specific code restores CPU context register. Only SMP case has the special code to restore the additional context:
https://github.com/apache/nuttx/blob/master/sched/irq/irq_csection.c#L717

> Three. The critical section is in oder to resolving some architecture can't disable global interrupt. But why other RTOS projects don't have this feature?
> 
> Finally, what is the difference between semaphore and enter_critical_section?

You can't use semaphore to protect the resource shared/accessed by irq handler, which mean you can call sem_post, but not sem_wait in the irq handler.

> What condition we should use semaphore instead of enter_critical_section? What condition we should use enter_critical_section instead of semaphore?

From the real time perspective, it's better to use semaphore as much as possible, but we found that semaphore is much slower than the critical section. So, we need find the balance here.

TaiJuWu, 2023-10-20:
> Yes, it should be replaced with spinlock. Actually, we have an issue track this: #1144.

So all functions are not called in IRQ should be replace with spinlock_irqsave?
Are there any other constraints?

xiaoxiang781216, 2023-10-20:
> So all functions are not called in IRQ should be replace with spinlock_irqsave? Are there any other constraints?

If the code inside critical section doesn't call functions which may block self or wake other, the critical section could be replaced with spinlock.

tmedicci, 2023-10-23:
Still about this question, I've recently reported [a problem with critical sections + pausing the other CPU (on SMP)](https://github.com/apache/nuttx/pull/10496#issuecomment-1762851740). Basically, it seems that we shouldn't call any function that calls `up_cpu_pause` in a critical section, otherwise, it'll hang...

That's another reason not to use `enter/leave_critical_section` without needing it.

closed 2023-10-29 by TaiJuWu