apache/nuttx #10988: Supported Platforms in Documentation

raiden00pl, 2023-10-21:
It seems that we have 2 separate places with `Supported Platforms`
One is here  [[1]](https://nuttx.apache.org/docs/latest/introduction/supported_platforms.html) and the other is here:  [[2]](https://nuttx.apache.org/docs/latest/platforms/index.html)

Both pages are incomplete.

[1] has not been updated for a long time and comments regarding the status seem very out of date.
I think we should move from [1] to [2] what is missing and remove [1]

TimJTi, 2023-10-21:
Not looked at the status details, but 1) is easier to use IMHO. A list of architectures/chips/SOCs with links to page(s) that lists the boards. Isn't that "neater"?

raiden00pl, 2023-10-21:
> Not looked at the status details, but 1) is easier to use IMHO. A list of architectures/chips/SOCs with links to page(s) that lists the boards. Isn't that "neater"?

I agree that 1 looks better. 
I think [2] is inspired by [Zephyr](https://docs.zephyrproject.org/latest/boards/index.html) and this layout it's not very readable.
But we can deal with visual issues later the outdated list is a bigger problem.

raiden00pl, 2023-10-23:
closed with https://github.com/apache/nuttx/pull/10991

closed 2023-10-23 by raiden00pl