apache/nuttx #11067: menuconfig : unable to modifier textbox based entry

WormMaster, 2023-10-27:
I tried to modify some parameters based on a textbox entry with the Menuconfig tool but I can't erase the default entry but I can add characters after the default entry. 

Tried on 2 different environments: 

- WSL2 with Ubuntu 22.04.3 LTS and the regular toolchain described in the official Nuttx documentation
- A VPS with the same OS and the same toolchain as mentioned previously

Tried with 2 different default configurations: 
- nucleo-f303re:nsh
- nucleo-f429zi:nsh

For exemple the USART3 serial console baudrate parameter : .config - NuttX/x86_64 Configuration
 → Device Drivers → Serial Driver Support → USART3 Configuration

acassis, 2023-10-27:
Are you pressing Space or Enter to toggle or enter in the field to edit it?
[image]

WormMaster, 2023-10-27:
I press enter on the field, but I can't erase the default entry (in your example 256 / 256 / 115200 values)

acassis, 2023-10-27:
Strange, backspace is not working? What are you using: kconfig-frontend or kconfiglib?

WormMaster, 2023-10-27:
Yes the backspace not working. I use kconfig-frontend

acassis, 2023-10-27:
Understood, please verify if kconfiglib works for you. I always used kconfig-frontend and it always worked on Linux and WSL, but I didn't test on WSL2 recently.

You can install kconfiglib this way:

$ pip install kconfiglib

WormMaster, 2023-10-28:
Thank you! It works with kconfiglib

closed 2023-10-28 by WormMaster